maocypher/Octoprint-Smart-Filament-Sensor #2: Does not work

Rhylthus, 2020-10-20:
Hi,

I have installed version 1.0.1 and it doesnt show up in octoprint plugins after installation. I'm on Python 3, so first version also doens't work. let me know if you need anithing to fix this. Thanks.

maocypher, 2020-10-22:
Hi, did you restart Octoprint, like it is describe in the installation manual?

Rhylthus, 2020-10-22:
Yes i have restart several time. I have also tried to install other plugin ans they work flawlessly. Any clue?

maocypher, 2020-10-22:
I try to reproduce it. Is there also Python 2.x installed on your machine or only Python 3.x?

Rhylthus, 2020-10-22:
Sorry i dont know. How can i check? Fyi first version was also not working.

maocypher, 2020-10-22:
On your linux terminal you can try
- python2 --version
- ls /usr/bin/python*

to name to alternatives.

Your exact python3 and octoprint version would also be helpful for me

Rhylthus, 2020-10-22:
Here you go 

OctoPrint version : 1.4.2
OctoPi version    : 0.17.0
------------------------------------------------------------------------------

python2 --version
Python 2.7.16
python3 --version
Python 3.7.3

Thanks.

maocypher, 2020-10-22:
Okay, I have the same python versions, but I installed Octoprint on Raspbian and I am not running it on OctoPi image. But it is running without errors.
Could you take a look into **<your installation path>/.octoprint/logs/octoprint.log** after restarting Octoprint if you can see any errors or provide me the log?
After start up it should start with the following lines

``` BASH
2020-10-22 15:47:27,549 - octoprint.startup - INFO - ******************************************************************************
2020-10-22 15:47:27,550 - octoprint.startup - INFO - Starting OctoPrint 1.4.2
2020-10-22 15:47:27,551 - octoprint.startup - INFO - ******************************************************************************
```

Rhylthus, 2020-10-22:
Here it is after a restart: seems bad?

2020-10-22 20:59:54,887 - octoprint.startup - INFO - ******************************************************************************
2020-10-22 20:59:54,889 - octoprint.startup - INFO - Starting OctoPrint 1.4.2
2020-10-22 20:59:54,894 - octoprint.startup - INFO - ******************************************************************************
2020-10-22 20:59:55,603 - octoprint.util.connectivity.connectivity_checker - INFO - Connectivity state is currently: offline
2020-10-22 20:59:55,603 - octoprint.util.connectivity.connectivity_checker - INFO - Connecting to 8.8.8.8:53 is not working
2020-10-22 20:59:55,603 - octoprint.util.connectivity.connectivity_checker - INFO - Resolving octoprint.org is not working
2020-10-22 20:59:55,611 - octoprint.startup - INFO - We don't appear to be online, not fetching plugin blacklist
2020-10-22 20:59:55,704 - octoprint.plugin.core - INFO - Loading plugins from /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins, /home/pi/.octoprint/plugins and installed plugin packages...
2020-10-22 20:59:58,210 - octoprint.plugin.core - ERROR - Error loading plugin bedlevelvisualizer
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/numpy/core/__init__.py", line 22, in <module>
    from . import multiarray
  File "/home/pi/oprint/lib/python3.7/site-packages/numpy/core/multiarray.py", line 12, in <module>
    from . import overrides
  File "/home/pi/oprint/lib/python3.7/site-packages/numpy/core/overrides.py", line 7, in <module>
    from numpy.core._multiarray_umath import (
ImportError: libf77blas.so.3: cannot open shared object file: No such file or directory

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/plugin/core.py", line 972, in _import_plugin
    module = _load_module(module_name, spec)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/plugin/core.py", line 71, in _load_module
    return imp.load_module(name, f, filename, details)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/vendor/imp.py", line 241, in load_module
    return load_package(name, filename)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/vendor/imp.py", line 215, in load_package
    return _load(spec)
  File "<frozen importlib._bootstrap>", line 696, in _load
  File "<frozen importlib._bootstrap>", line 677, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 728, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_bedlevelvisualizer/__init__.py", line 7, in <module>
    import numpy as np
  File "/home/pi/oprint/lib/python3.7/site-packages/numpy/__init__.py", line 140, in <module>
    from . import core
  File "/home/pi/oprint/lib/python3.7/site-packages/numpy/core/__init__.py", line 48, in <module>
    raise ImportError(msg)
ImportError: 

IMPORTANT: PLEASE READ THIS FOR ADVICE ON HOW TO SOLVE THIS ISSUE!

Importing the numpy C-extensions failed. This error can happen for
many reasons, often due to issues with your setup or how NumPy was
installed.

We have compiled some common reasons and troubleshooting tips at:

    https://numpy.org/devdocs/user/troubleshooting-importerror.html

Please note and check the following:

  * The Python version is: Python3.7 from "/home/pi/oprint/bin/python"
  * The NumPy version is: "1.19.2"

and make sure that they are the versions you expect.
Please carefully study the documentation linked above for further help.

Original error was: libf77blas.so.3: cannot open shared object file: No such file or directory

2020-10-22 20:59:58,390 - octoprint.plugins.octoprint.plugins.discovery - INFO - pybonjour is not installed, Zeroconf Discovery won't be available
2020-10-22 20:59:58,413 - octoprint.plugin.core - INFO - Found 29 plugin(s) providing 29 mixin implementations, 57 hook handlers
2020-10-22 20:59:58,677 - octoprint.server.heartbeat - INFO - Starting server heartbeat, 900.0s interval
2020-10-22 20:59:58,722 - octoprint.server - INFO - Intermediary server started
2020-10-22 20:59:58,723 - octoprint.plugin.core - INFO - Loading plugins from /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins, /home/pi/.octoprint/plugins and installed plugin packages...
2020-10-22 20:59:58,868 - octoprint.plugin.core - ERROR - Error loading plugin bedlevelvisualizer
Traceback (most recent call last):
[... 145 lines cut ...]
2020-10-22 21:00:05,984 - octoprint.plugins.pluginmanager - INFO - Loaded plugin repository data from disk, was still valid
2020-10-22 21:00:07,114 - octoprint.util.pip - INFO - Using "/home/pi/oprint/bin/python -m pip" as command to invoke pip
2020-10-22 21:00:07,131 - octoprint.util.pip - INFO - pip installs to /home/pi/oprint/lib/python3.7/site-packages (writable -> yes), --user flag needed -> no, virtual env -> yes
2020-10-22 21:00:07,131 - octoprint.util.pip - INFO - ==> pip ok -> yes
2020-10-22 21:00:07,163 - octoprint.plugins.softwareupdate - INFO - Minimum free storage across all update relevant locations is 8.5GB. That is considered sufficient for updating.
2020-10-22 21:00:07,247 - octoprint.util.connectivity.connectivity_checker - INFO - Connectivity changed from offline to online
[... cut ...]
2020-10-22 21:00:11,870 - octoprint.util.comm - INFO - Changing monitoring state from "Detecting serial connection" to "Error: No more candidates to test, and no working port/baudrate combination detected."
2020-10-22 21:00:11,877 - octoprint.plugins.smartfilamentsensor - INFO - Error: Disabling filament sensors.
2020-10-22 21:00:11,879 - octoprint.util.comm - INFO - Changing monitoring state from "Error: No more candidates to test, and no working port/baudrate combination detected." to "Offline (Error: No more candidates to test, and no working port/baudrate combination detected.)"
2020-10-22 21:00:11,893 - octoprint.plugins.action_command_notification - INFO - Notifications cleared

Rhylthus, 2020-10-23:
I have re-install on a fresh sd card octoprint and this plugin and everithing is working. So it must be something messing from my side. Thanks for your help.

closed 2020-10-23 by Rhylthus