maocypher/Octoprint-Smart-Filament-Sensor #4: wich port/pin to use

wuast94, 2020-10-22:
What do you recommend to use? 

https://www.raspberrypi-spy.co.uk/wp-content/uploads/2012/06/Raspberry-Pi-GPIO-Header-with-Photo.png

or are all GPIO pins possible ?

maocypher, 2020-10-23:
Hi :) from this I used the free green GPIO pins e.g. 11 in Board Mode (Pin number) i.e. GPIO 17 in BCM Mode.
Other GPIO pins like GPIO 2 are for special usage (in this case I2C) which might be necessary for some sensors you probably want to attach to your Raspberry.

Take care to use VCC with **3.3V**, because GPIO pins don't last very long with 5V.

maocypher, 2020-10-25:
I added this to the README.MD and will close this issue tomorrow, if this answered your question.

wuast94, 2020-10-27:
Sry for the late reply.. Don't seen it yet.. But yes that answered my question and it's nice to add it to the Doku :) 

Tanks :)

maocypher, 2020-10-31:
Recommendation is put into description how to use the plugin

closed 2020-10-31 by maocypher

liebig, 2020-11-25:
Sorry for the stupid question, but how do I wire the Bigtreetech Filament Sensor to the Raspberry Pi?  The cable consists of a 4 pole end piece, whereby only 3 wires are connected here. Can you tell me which cable is used for power, ground and data? This might also be something for the documentary.

Thanks a lot!

maocypher, 2020-11-25:
There are no stupid questions: https://github.com/bigtreetech/smart-filament-detection-module/tree/master/manual
On page 12 of the manual you can find which pin is for what usage. Please check if the wiring on the board side of the cable is correct. I heard there are some cables with pole twists.

VCC connect to 3.3V
SIN to GPIO 11, if you know how to use others feel free
GND to GND

liebig, 2020-11-25:
Wow thank you very much for the quick response. I will test this. Thanks for your great work here!