maocypher/Octoprint-Smart-Filament-Sensor #5: Starting Purge Line Issues

sushibagel, 2020-10-27:
First of all thanks for your hard work! The plugin works as expected for the most part. The only issue I'm having is it keeps pausing my print after the purge line. I left the default detection time on (45 Seconds) in the settings, I'm thinking it's stopping because of the way my start code is for my ABL probe (I only partially heat my nozzle, probe, and then finish heating) and I believe it just happens to let the purge line go and then pauses. (I could be wrong) 

I think a simple solution may be to add an ignore XX seconds/minutes at the beginning of the print, this will allow the print to actually get started before the sensor starts monitoring for a runout/jam/etc.

maocypher, 2020-10-28:
Hello yes with ABL probe the Z-Axis is changed which is my starting point. A delay is a possible solution for this.

If you could provide me the start part of your gcode, I could analyse it to get a better starting sequence :)

For the time being I didn't take a closer look on ABL.

sushibagel, 2020-10-28:
Thanks so much for looking into this. Here's my starting G-Code: 

`; Ender 3 Custom Start G-code
M140 S{material_bed_temperature_layer_0} ; Set Heat Bed temperature
M190 S{material_bed_temperature_layer_0} ; Wait for Heat Bed temperature
M104 S160; start warming extruder to 160
G28 ; Home all axes
@BEDLEVELVISUALIZER	; instruct plugin to start recording responses from printer.
M117 Generating mesh...
G29 T; Auto bed-level
G92 E0 ; Reset Extruder
M104 S{material_print_temperature_layer_0} ; Set Extruder temperature
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
M109 S{material_print_temperature_layer_0} ; Wait for Extruder temperature
G1 X0.1 Y20 Z0.3 F5000.0 ; Move to start position
; G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X0.1 Y200.0 Z0.3 F1500.0 E15 ; Draw the first line
G1 X0.4 Y200.0 Z0.3 F5000.0 ; Move to side a little
G1 X0.4 Y20 Z0.3 F1500.0 E30 ; Draw the second line
G92 E0 ; Reset Extruder
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
; End of custom start GCode`

maocypher, 2020-10-28:
Thanks for the GCode.
Short question: I noticed that you run
```
G29 T; Auto bed-level
G92 E0 ; Reset Extruder
M104 S{material_print_temperature_layer_0} ; Set Extruder temperature
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
M109 S{material_print_temperature_layer_0} ; Wait for Extruder temperature
G1 X0.1 Y20 Z0.3 F5000.0 ; Move to start position

```
Are you heating up again on M109 after G1 so that time is passing?
Since there is no real movement after G29 for ABL you could try to finalize heating up completly, before running G1 command to move the nozzle upwards to not crash the heatbed.
So, swapping these two lines could solve your problem.

Could you try this start code?

```
; Ender 3 Custom Start G-code
M140 S{material_bed_temperature_layer_0} ; Set Heat Bed temperature
M190 S{material_bed_temperature_layer_0} ; Wait for Heat Bed temperature
M104 S160; start warming extruder to 160
G28 ; Home all axes
@BEDLEVELVISUALIZER ; instruct plugin to start recording responses from printer. M117 Generating mesh...
G29 T; Auto bed-level
G92 E0 ; Reset Extruder
;Heat up completely before the first movement (G1 command)
M104 S{material_print_temperature_layer_0} ; Set Extruder temperature
M109 S{material_print_temperature_layer_0} ; Wait for Extruder temperature
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X0.1 Y20 Z0.3 F5000.0 ; Move to start position ;
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X0.1 Y200.0 Z0.3 F1500.0 E15 ; Draw the first line
G1 X0.4 Y200.0 Z0.3 F5000.0 ; Move to side a little
G1 X0.4 Y20 Z0.3 F1500.0 E30 ; Draw the second line
G92 E0 ; Reset Extruder
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
; End of custom start GCode
```

sushibagel, 2020-10-29:
Ok I'll try and give this a try, having an unrelated issue with my printer right now that I'll need to solve before I can test this

sushibagel, 2020-10-29:
Just reporting back. I tried the start code you suggested and it still stopped shortly after the print began.

Rhylthus, 2020-10-29:
Have the same issue (with same starting g-code). I have put a default detection time of 120 seconds and it seems to work, print is continuing. So i guess it's confirmed that issue is related to starting g-code.

maocypher, 2020-10-29:
Thank you for giving this feedbsck. It will help me analyse the problem.

I try to provide a bugfix for it :)

If you like, I could provide a pre-release version for you, where you could try if the bugfix is working. Because I don't have ABL and can't test it in short term.

sushibagel, 2020-10-29:
> Thank you for giving this feedbsck. It will help me analyse the problem.
> 
> I try to provide a bugfix for it :)
> 
> If you like, I could provide a pre-release version for you, where you could try if the bugfix is working. Because I don't have ABL and can't test it in short term.

Thanks! I'd be happy to test.

Rhylthus, 2020-10-29:
> Thank you for giving this feedbsck. It will help me analyse the problem.
> 
> I try to provide a bugfix for it :)
> 
> If you like, I could provide a pre-release version for you, where you could try if the bugfix is working. Because I don't have ABL and can't test it in short term.

yes of course thanks !

Rhylthus, 2020-10-30:
Is-it possible to see in the log or elsewere the signal read from the Pi from the sensor? I would like to be sure that everything is connected and working properly.

Thanks.

sushibagel, 2020-10-30:
> Is-it possible to see in the log or elsewere the signal read from the Pi from the sensor? I would like to be sure that everything is connected and working properly.
> 
> Thanks.

I'd be happy to provide them, just not sure where to look.

maocypher, 2020-10-30:
You can find the logs in *<your installation path>/.octoprint/logs/octoprint.log*
When the sensor starts there is a logline
*motion sensor started*
If you like to see more logs. I could add more logging again, to see every sensor signal, but it flodded the log file. And log level Debug was not working as expected

Rhylthus, 2020-10-30:
> You can find the logs in _/.octoprint/logs/octoprint.log_
> When the sensor starts there is a logline
> _motion sensor started_
> If you like to see more logs. I could add more logging again, to see every sensor signal, but it flodded the log file. And log level Debug was not working as expected

If possible i would like to have the full log in order to see if everithing is working and connected properly. If you can give me a special build or if you can add a toggle in the plugin it would be helpfull. I dont want you to work on something if the issue is on my side. Thanks.

maocypher, 2020-10-30:
I created a new branch for this issue: ABL-Support
You can download the version with enhanced logging here: https://github.com/maocypher/Octoprint-Smart-Filament-Sensor/archive/ABL-Support.zip
In Settings -> Logging you must add DEBUG Level for octoprint.plugins.smartfilamentsensor

After you tried this you can grep the octoprint.log with this command and upload it here.
``` BASH
grep -E 'octoprint\.plugins\.smartfilamentsensor' octoprint.log > smart-filament-sensor.log
```

I tried Start GCode without G1 or G0 commands only with heating up, homing G28 and ABL G29, but there was nothing in my logs to see, i.e. the sensor did not start as expected. If you could also try this and provide me the logs, it would be very helpful.

For your information: The starting event for the sensor is the first G1 or G0 command that is processed by the printer.

If you are not sure, if you connected the sensor correctly you could check the used GPIO pin again and the used GPIO mode https://www.raspberrypi.org/documentation/usage/gpio/
_GPIO Board Mode = the number of the pin
GPIO BCM Mode = the GPIO number_

To check if your sensor is properly connected you could run this python script. Therefore only the sensor must be connected to the Pi you can move a piece of filament by hand. To run it successfully you must change USED_PIN value and set your Board Mode to the one you want to use.

``` Python
#!/usr/bin/python3
import RPi.GPIO as GPIO
import time

#CONST
USED_PIN = 11
GPIO.setmode(GPIO.BOARD)
#GPIO.setmode(GPIO.BCM)
# Time in seconds
max_not_moving_time = 2
# Set up the GPIO channels - one input and one output
GPIO.setup(USED_PIN, GPIO.IN)

lastValue = GPIO.input(USED_PIN)
# Get current time in seconds
lastMotion = time.time()

def main():
	try: 
		GPIO.add_event_detect(USED_PIN, GPIO.BOTH, callback=motion)

		while True:
			timespan = (time.time() - lastMotion)

			if (timespan > max_not_moving_time):
				print("No motion detected")
			else:
				print ("Moving")

			time.sleep(0.250)
		
		GPIO.remove_event_detect(USED_PIN)
	except KeyboardInterrupt:
		print ("Done")
		pass

def motion(pPin):
	global lastMotion
	lastMotion = time.time()
	print("Motion detected at " + str(lastMotion))

main()
```

Rhylthus, 2020-10-30:
> I created a new branch for this issue: ABL-Support
> You can download the version with enhanced logging here: https://github.com/maocypher/Octoprint-Smart-Filament-Sensor/archive/ABL-Support.zip
> In Settings -> Logging you must add DEBUG Level for octoprint.plugins.smartfilamentsensor
> 
> After you tried this you can grep the octoprint.log with this command and upload it here.
> 
> ```shell
> grep -E 'octoprint\.plugins\.smartfilamentsensor' octoprint.log > smart-filament-sensor.log
> ```
> 
> I tried Start GCode without G1 or G0 commands only with heating up, homing G28 and ABL G29, but there was nothing in my logs to see, i.e. the sensor did not start as expected. If you could also try this and provide me the logs, it would be very helpful.
> 
> For your information: The starting event for the sensor is the first G1 or G0 command that is processed by the printer.
> 
> If you are not sure, if you connected the sensor correctly you could check the used GPIO pin again and the used GPIO mode https://www.raspberrypi.org/documentation/usage/gpio/
> _GPIO Board Mode = the number of the pin GPIO BCM Mode = the GPIO number_
> 
> To check if your sensor is properly connected you could run this python script. Therefore only the sensor must be connected to the Pi you can move a piece of filament by hand. To run it successfully you must change USED_PIN value and set your Board Mode to the one you want to use.
> 
> ```python
> #!/usr/bin/python3
> import RPi.GPIO as GPIO
> import time
> 
> #CONST
> USED_PIN = 11
> GPIO.setmode(GPIO.BOARD)
> #GPIO.setmode(GPIO.BCM)
> # Time in seconds
> max_not_moving_time = 2
> # Set up the GPIO channels - one input and one output
> GPIO.setup(USED_PIN, GPIO.IN)
> 
> lastValue = GPIO.input(USED_PIN)
> # Get current time in seconds
> lastMotion = time.time()
> 
> def main():
> 	try: 
> 		GPIO.add_event_detect(USED_PIN, GPIO.BOTH, callback=motion)
> 
> 		while True:
> 			timespan = (time.time() - lastMotion)
> 
> 			if (timespan > max_not_moving_time):
> 				print("No motion detected")
> 			else:
> 				print ("Moving")
> 
> 			time.sleep(0.250)
> 		
> 		GPIO.remove_event_detect(USED_PIN)
> 	except KeyboardInterrupt:
> 		print ("Done")
> 		pass
> 
> 
> def motion(pPin):
> 	global lastMotion
> 	lastMotion = time.time()
> 	print("Motion detected at " + str(lastMotion))
> 
> main()
> ```

Thanks !!! With you script everything is working now. I have made a mistake when plug (G and V). I will try again with my initial stating g-code tomorrow to check if it's working with ABL.

thanks again.

sushibagel, 2020-10-31:
> I created a new branch for this issue: ABL-Support
> You can download the version with enhanced logging here: https://github.com/maocypher/Octoprint-Smart-Filament-Sensor/archive/ABL-Support.zip
> In [...]

I may owe you an apology. I tested the new version but still had the issue. But on the same print I happened to run out if filament, waited and waited but the print didn't stop so I must have installed my sensor incorrectly... I can't troubleshoot the sensor install at the moment but will give it a try this evening.

What's strange is that I tested the runout situation when I first installed and it seemed to work but now I'm wondering if it was just happening to stop at the point I was cutting the filament because of whatever is stopping my prints at the start-ish now... I don't really know but I'll report back one I screw with it some more.

sushibagel, 2020-10-31:
Ok, so pins are connected correctly and I ran the python script and confirmed movement is being detected. I will try another print and report back.

sushibagel, 2020-10-31:
> Ok, so pins are connected correctly and I ran the python script and confirmed movement is being detected. I will try another print and report back.

Tried another quick print. Start worked fine now but about cut the filament about 5 minutes in, let it finish printing (about 10 minutes) and it didn't pause. Not sure where to go from here...

maocypher, 2020-10-31:
> Tried another quick print. Start worked fine now but about cut the filament about 5 minutes in, let it finish printing (about 10 minutes) and it didn't pause. Not sure where to go from here...

First of all, that the print interrupts after cutting is good. Testing the jam situation with 45sec timeout is quite hard, but it is the same logic like with filament run out.

The last situation to check if it is finishing a print on its own. So, without any manipulation from your side. I alwaya took a X-Y-Z cube for this. Did this also work?

maocypher, 2020-10-31:
> Thanks !!! With you script everything is working now. I have made a mistake when plug (G and V). I will try again with my initial stating g-code tomorrow to check if it's working with ABL.
>
> thanks [...]

Great to hear :) let me know if it is also working with ABL.

You could check with your stating code and once with mine.
If you are unsure about finally heating the nozzle very near to the heatbed you could call auto homing before.

sushibagel, 2020-10-31:
> First of all, that the print interrupts after cutting is good. Testing the jam situation with 45sec timeout is quite hard, but it is the same logic like with filament run out.
>
> The last situation to [...]

Prints finish just fine but now the sensor isn't detecting a runout. When I did the cut test it just kept going till the print was finished.

maocypher, 2020-10-31:
Okay, is the configuration in the Plugin the same like in the script you used?

Did you take a look into the log, that the sensor even started or ran into an error?

In a previous comment I sent a command for filtering.

Sometimes restarting the Octoprint helped for me, when I was searching for errors

sushibagel, 2020-10-31:
> Okay, is the configuration in the Plugin the same like in the script you used?
>
> Did you take a look into the log, that the sensor even started or ran into an error?
>
> In a previous comment I sent a [...]

It does seem to have started... 

`2020-10-30 22:23:08,223 - octoprint.plugins.smartfilamentsensor - DEBUG - Sensor enabled: True
2020-10-30 22:23:08,224 - octoprint.plugins.smartfilamentsensor - DEBUG - GPIO mode: Board Mode
2020-10-30 22:23:08,226 - octoprint.plugins.smartfilamentsensor - DEBUG - GPIO pin: 11
2020-10-30 22:23:08,226 - octoprint.plugins.smartfilamentsensor - DEBUG - Detection Mode: Timeout detection
2020-10-30 22:23:08,228 - octoprint.plugins.smartfilamentsensor - DEBUG - Timeout: 45
2020-10-30 22:23:08,264 - octoprint.plugins.smartfilamentsensor - INFO - Motion sensor started: Timeout detection`

maocypher, 2020-10-31:
Yes, looks fine so far.
Did you connect it to 5V or 3.3V as described?

sushibagel, 2020-10-31:
> Yes, looks fine so far.
> Did you connect it to 5V or 3.3V as described?

3.3v

maocypher, 2020-10-31:
Another thing you could try:
Enable VIRTUAL printer and run your print there. The filament is not moving in the sensor. I.e. after 45sec the print is paused. For testing you can also set this value down to 10secs for example.

Is M600 still configured as Pause-Event for Octoprint?

sushibagel, 2020-10-31:
Pausing commands under settings "Firmware Protocols" are M0, M1, M25 

I'll give a virtual printer a try in the morning.

maocypher, 2020-10-31:
You must add M600 as pausing command, otherwise Octoprint ignores this command.

> Since currently GCode command M600 is used to interrupt the print, it is recommended to add M600 to the setting “Pausing commands”.

sushibagel, 2020-10-31:
> You must add M600 as pausing command, otherwise Octoprint ignores this command.
> 
> > Since currently GCode command M600 is used to interrupt the print, it is recommended to add M600 to the setting “Pausing commands”.

Ahh I missed that part in the instructions 🤦‍♂️I'll add that and try again tomorrow. Thanks

sushibagel, 2020-10-31:
> You must add M600 as pausing command, otherwise Octoprint ignores this command.
> 
> > Since currently GCode command M600 is used to interrupt the print, it is recommended to add M600 to the setting “Pausing commands”.

I'm getting pretty frustrated now. Not sure what I'm doing wrong. I put in the M600, "Serial Connection > Firmware & Protocol > Pausing Commands" - first print worked as expected, second print kept pausing not sure why.

maocypher, 2020-10-31:
Can you provide me the log of the second print, so that I can try to help you?

sushibagel, 2020-10-31:
> Can you provide me the log of the second print, so that I can try to help you?

Thanks for your continued help... I believe this is the relevant part. 

`2020-10-31 10:59:27,732 - octoprint.plugins.smartfilamentsensor - DEBUG - Sensor enabled: True
2020-10-31 10:59:27,739 - octoprint.plugins.smartfilamentsensor - DEBUG - GPIO mode: Board Mode
2020-10-31 10:59:27,759 - octoprint.plugins.smartfilamentsensor - DEBUG - GPIO pin: 11
2020-10-31 10:59:27,766 - octoprint.plugins.smartfilamentsensor - DEBUG - Detection Mode: Timeout detection
2020-10-31 10:59:27,775 - octoprint.plugins.smartfilamentsensor - DEBUG - Timeout: 45
2020-10-31 10:59:27,823 - octoprint.plugins.smartfilamentsensor - INFO - Motion sensor started: Timeout detection
2020-10-31 11:00:12,880 - octoprint.plugins.smartfilamentsensor - DEBUG - Motion sensor detected no movement
2020-10-31 11:00:12,961 - octoprint.util.comm - INFO - Pausing print job due to command M600
2020-10-31 11:00:12,963 - octoprint.util.comm - INFO - Changing monitoring state from "Printing" to "Pausing"
2020-10-31 11:01:03,159 - octoprint.printer.standard.job - INFO - Print job paused - origin: local, path: Camera_Mount_6.5_Hours.gcode, owner: Drew, user: None
2020-10-31 11:01:03,181 - octoprint.plugins.smartfilamentsensor - INFO - Motion sensor stopped
2020-10-31 11:01:03,192 - octoprint.util.comm - INFO - Changing monitoring state from "Pausing" to "Paused"
2020-10-31 11:01:03,202 - octoprint.plugins.telegram - INFO - Sending a message: Paused printing Camera_Mount_6.5_Hours.gcode at 0%. 11:11:49 remaining and the print head is currently located at 0.2. with_image=True with_gif=0 chatID= 1380430573
2020-10-31 11:01:12,727 - octoprint.util.comm - INFO - Changing monitoring state from "Paused" to "Resuming"
2020-10-31 11:01:12,734 - octoprint.printer.standard.job - INFO - Print job resumed - origin: local, path: Camera_Mount_6.5_Hours.gcode, owner: Drew, user: Drew
2020-10-31 11:01:12,755 - octoprint.util.comm - INFO - Changing monitoring state from "Resuming" to "Printing"
2020-10-31 11:01:12,806 - octoprint.plugins.smartfilamentsensor - DEBUG - Sensor enabled: True
2020-10-31 11:01:12,808 - octoprint.plugins.smartfilamentsensor - DEBUG - GPIO mode: Board Mode
2020-10-31 11:01:12,812 - octoprint.plugins.smartfilamentsensor - DEBUG - GPIO pin: 11
2020-10-31 11:01:12,818 - octoprint.plugins.smartfilamentsensor - DEBUG - Detection Mode: Timeout detection
2020-10-31 11:01:12,822 - octoprint.plugins.smartfilamentsensor - DEBUG - Timeout: 45
2020-10-31 11:01:12,861 - octoprint.plugins.smartfilamentsensor - INFO - Motion sensor started: Timeout detection
2020-10-31 11:01:12,871 - octoprint.plugins.telegram - INFO - Sending a message: Resumed printing Camera_Mount_6.5_Hours.gcode at 0%. 11:11:49 remaining. with_image=True with_gif=0 chatID= 1380430573
2020-10-31 11:01:57,934 - octoprint.plugins.smartfilamentsensor - DEBUG - Motion sensor detected no movement
2020-10-31 11:01:57,951 - octoprint.util.comm - INFO - Pausing print job due to command M600
2020-10-31 11:01:57,953 - octoprint.util.comm - INFO - Changing monitoring state from "Printing" to "Pausing"
2020-10-31 11:02:01,829 - octoprint.printer.standard.job - INFO - Print job paused - origin: local, path: Camera_Mount_6.5_Hours.gcode, owner: Drew, user: None
2020-10-31 11:02:01,850 - octoprint.plugins.smartfilamentsensor - INFO - Motion sensor stopped
2020-10-31 11:02:01,871 - octoprint.plugins.telegram - INFO - Sending a message: Paused printing Camera_Mount_6.5_Hours.gcode at 0%. 11:11:18 remaining and the print head is currently located at 0.2. with_image=True with_gif=0 chatID= 1380430573
2020-10-31 11:02:42,729 - octoprint.util.comm - INFO - Changing monitoring state from "Pausing" to "Paused"
2020-10-31 11:02:58,466 - octoprint.util.comm - INFO - Changing monitoring state from "Paused" to "Cancelling"
2020-10-31 11:02:58,476 - octoprint.plugins.excluderegion - INFO - Printing stopped: event=PrintCancelling
2020-10-31 11:02:59,141 - octoprint.printer.standard.job - INFO - Print job cancelled - origin: local, path: Camera_Mount_6.5_Hours.gcode, owner: Drew, user: Drew
2020-10-31 11:02:59,146 - octoprint.plugins.DisplayLayerProgress - INFO - Printing stopped. Detailed progress stopped.
2020-10-31 11:02:59,176 - octoprint.plugins.SpoolManager - INFO - Extruded filament length: 76.19171
2020-10-31 11:02:59,176 - octoprint.plugins.SpoolManager - INFO - Current Spool filament length: 215616
2020-10-31 11:02:59,178 - octoprint.plugins.SpoolManager - INFO - New Spool filament length: 215692.19171
2020-10-31 11:02:59,218 - octoprint.plugins.detailedprogress - INFO - Printing stopped. Detailed progress stopped.
2020-10-31 11:02:59,227 - octoprint.plugins.excluderegion - INFO - Printing stopped: event=PrintCancelled
2020-10-31 11:02:59,229 - octoprint.plugins.smartfilamentsensor - INFO - PrintCancelled: Disabling filament sensors.
2020-10-31 11:02:59,323 - octoprint.util.comm - INFO - Changing monitoring state from "Cancelling" to "Operational"
2020-10-31 11:02:59,342 - octoprint.plugins.DisplayLayerProgress - INFO - Printing stopped. Detailed progress stopped.
2020-10-31 11:02:59,349 - octoprint.plugins.detailedprogress - INFO - Printing stopped. Detailed progress stopped.
2020-10-31 11:02:59,356 - octoprint.plugins.excluderegion - INFO - Printing stopped: event=PrintFailed
2020-10-31 11:02:59,356 - octoprint.plugins.smartfilamentsensor - INFO - PrintFailed: Disabling filament sensors.
[... cut ...]
  File "/home/pi/oprint/local/lib/python2.7/site-packages/tornado/http1connection.py", line 451, in write
    self._pending_write = self.stream.write(self._format_chunk(chunk))
  File "/home/pi/oprint/local/lib/python2.7/site-packages/tornado/http1connection.py", line 424, in _format_chunk
    "Tried to write more data than Content-Length")
HTTPOutputError: Tried to write more data than Content-Length
2020-10-31 11:09:46,375 - tornado.general - ERROR - Cannot send error response after headers written`

Rhylthus, 2020-11-01:
i have tested. With my starting G-code it paused the print at the begining.

; Ender 3 Custom Start G-code
M140 S{material_bed_temperature_layer_0} ; Set Heat Bed temperature
M190 S{material_bed_temperature_layer_0} ; Wait for Heat Bed temperature
M104 S160; start warming extruder to 160
G28 ; Home all axes
G29 ; Auto bed-level (BL-Touch)
G92 E0 ; Reset Extruder
M104 S{material_print_temperature_layer_0} ; Set Extruder temperature
G1 X0.1 Y20 Z0.3 F5000.0 ; Move to start position
M109 S{material_print_temperature_layer_0} ; Wait for Extruder temperature
; G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X0.4 Y200.0 Z0.3 F5000.0 ; Move to side a little
G1 X0.4 Y20 Z0.3 F1500.0 E30 ; Draw the second line
G92 E0 ; Reset Extruder
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
; End of custom start GCode

with this G-code it works:
; Ender 3 Custom Start G-code
M140 S{material_bed_temperature_layer_0} ; Set Heat Bed temperature
M190 S{material_bed_temperature_layer_0} ; Wait for Heat Bed temperature
M104 S160; start warming extruder to 160
G28 ; Home all axes
G29 ; Auto bed-level (BL-Touch)
G92 E0 ; Reset Extruder
M104 S{material_print_temperature_layer_0} ; Set Extruder temperature
M109 S{material_print_temperature_layer_0} ; Wait for Extruder temperature
G1 X0.1 Y20 Z0.3 F5000.0 ; Move to start position
; G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X0.4 Y200.0 Z0.3 F5000.0 ; Move to side a little
G1 X0.4 Y20 Z0.3 F1500.0 E30 ; Draw the second line
G92 E0 ; Reset Extruder
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
; End of custom start GCode

So you should precise in the readme to put the starting g-code G1 after pre-heating or just put a ignoring timer at the begining of the print.

Thanks for your help.

maocypher, 2020-11-01:
> Thanks for your continued help... I believe this is the relevant part.

From what I can see the sensor really didn't detect any movement? Did you check if the printer is jammed or just extruding very slowly? I had it once that after heating up again the printer did not extrude enough, so that it stopped every 45secs.
Did you also check the wiring that no wire slipped out?

maocypher, 2020-11-01:
> So you should precise in the readme to put the starting g-code G1 after pre-heating or just put a ignoring timer at the begining of the print.

Thank you very much for the feedback. I will add this to the description and put your starting GCode as an example!

maocypher, 2020-11-01:
@sushibagel I improved the Distance Detection Mode a little more. If you would like you could give this version a try
https://github.com/maocypher/Octoprint-Smart-Filament-Sensor/archive/M114-detection.zip

I also tested it with several prints in a row and it is working without issues (so far).

I would be glad about some feedback. @Rhylthus you are also invited to test it ;)

sushibagel, 2020-11-01:
Went through my system yesterday and reseated all the connections, adjusted the mount and did a system tune all seems to be working fine so far (my extruder may have been slipping). Haven't tested a runout situation yet but will try and do that later today.

Thanks for the update I'll get that loaded and give it a try too.

sushibagel, 2020-11-01:
Tried to test a runout situation twice waited a solid couple minutes to see if the print paused but nothing.

maocypher, 2020-11-01:
If you are still running in Debug Mode, this can happen, because then a lot is logged now.

Did you check if everything is configured properly?

Pausing should happen after a couple of seconds if the filament runs out.
I tested this situation in various combinations and it worked every time as expected.

If it is not pausing it could be:
- Sensor is disabled
- Value is too high, i.e. detection takes long time
- M600 not added as Pausing Command
- Instead of Distance Detection, Timeout detection with a high value
- Your GCode is different than expected
- Any exeption occured

sushibagel, 2020-11-01:
> If you are still running in Debug Mode, this can happen, because then a lot is logged now.
>
> Did you check if everything is configured properly?
>
> Pausing should happen after a couple of seconds if [...]

I did verify the plugin was on and M600 is still in as a pausing command. I was able to briefly open the log and did see it had logged movement. 
I have Timeout detection as 45 seconds. 

What sort of GCode difference would I look for?

sushibagel, 2020-11-01:
I am on the latest Octoprint version 1.4.2

maocypher, 2020-11-01:
Just send me the first lines of your GCode, I can analyse this.

Please use _Distance Detection_ with 15mm instead of _Timeout Detection_, it is be more reliable.
To stop producing tons of logs switch back to Loglevel **Info**

sushibagel, 2020-11-01:
Here are the first few lines below:

It's strange, not sure if this was a mistake but I updated the settings to be 15mm Distance Detection and my print paused right away and now keeps pausing. 

`;FLAVOR:Marlin
;TIME:78586
;Filament used: 56.3697m
;Layer height: 0.2
;MINX:3.937
;MINY:5.926
;MINZ:0.2
;MAXX:218.912
;MAXY:165.475
;MAXZ:60
;Generated with Cura_SteamEngine 4.6.2
M82 ;absolute extrusion mode
; Ender 3 Custom Start G-code
M140 S55.0 ; Set Heat Bed temperature
M190 S55.0 ; Wait for Heat Bed temperature
M104 S160; start warming extruder to 160
G28 ; Home all axes
@BEDLEVELVISUALIZER ; instruct plugin to start recording responses from printer.
M117 ;Generating mesh...
G29 T; Auto bed-level
G92 E0 ; Reset Extruder
;Heat up completely before the first movement (G1 command)
M104 S195.0 ; Set Extruder temperature
M109 S195.0 ; Wait for Extruder temperature
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X0.1 Y20 Z0.3 F5000.0 ; Move to start position ;
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X0.1 Y200.0 Z0.3 F1500.0 E15 ; Draw the first line
G1 X0.4 Y200.0 Z0.3 F5000.0 ; Move to side a little
G1 X0.4 Y20 Z0.3 F1500.0 E30 ; Draw the second line
G92 E0 ; Reset Extruder
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
; End of custom start GCode
G92 E0
G92 E0
G1 F3000 E-1
;LAYER_COUNT:300
;LAYER:0
M117 INDICATOR-Layer0
M107
G0 F6000 X9.008 Y96.397 Z0.2
;TYPE:SKIRT`

maocypher, 2020-11-01:
Try a value of 30mm as experiment, 15mm was more than fine for me.
If it is still continously pausing, than no signal was sent from your sensor.

You could try restarting your octoprint or even the raspberry. If there is still no signal from the sensor, although everything is wired correct, it could be a sensor problem.

sushibagel, 2020-11-01:
Gosh I hope this isn't a sensor issue.... this was my first ever purchase on AliExpress so I doubt I'm getting my money back or a replacement... But restarted my print with a 30mm distance, it's printing fine, but no runout detection

sushibagel, 2020-11-01:
Ran the script you provided earlier to see if the movement is being detected while my print was paused. It's detecting movement even with the print paused it does have some no movement but it's mostly showing movement. Is it picking up vibration from my printer maybe?

maocypher, 2020-11-02:
With the python script I provided you? If there is no movement, but the sensor is sending signals, this is more than strange. Did you choose a free GPIO or maybe one for special usage like CLK?
If you chose a free GPIO and there is definately no movement, but you fear that it could be vibrations - which I doubt - switch of the printer completly and only run the sensor without movement on the raspberry.

But as this seems to be another issue and related to your hardware, it makes more sense to close this bug.

I created a Slack Channel [#octoprint-smart-filament-sensor-plugin](https://maocypher.slack.com/archives/C01DG7C82UW) you could join it and I could try to give you some support there as far as it is possible for me. If the channel is not public yet, let me know and I try to figure out how to make it public

sushibagel, 2020-11-02:
Hi Maocypher,

I'll go ahead and give that stuff a try. I'd be happy to join the slack channel once you make it public so I can update you on what happens. For the time being, I'll close this bug report.

closed 2020-11-02 by sushibagel

sushibagel, 2020-11-03:
@maocypher Turns out I had my wires coming out of the wire harness wrong... The original diagram I looked at must have been wrong. But thanks again for all the help. Got a paypal or anything? I'd love to send a donation your way for all the work and help

maocypher, 2020-11-03:
@sushibagel Thank you very much for your feedback! I am happy that it is working now and you like my work!

For donation you could use this button:
[image]