maocypher/Octoprint-Smart-Filament-Sensor #6: Request: Sensor Test

xrxca, 2020-11-04:
I've been experimenting with this and feel it would be nice to have a sensor test button of some type.  I had a break in the cable going to the sensor so nothing was working, a test button would (maybe) have saved me a bit of time.
(click Test, Dialog pops up: "Move filament in sensor", then when movement is detected changes to "Movement Detected", or something along those lines.)

maocypher, 2020-11-04:
Hi, I am already working on this. Till it is finished I am going to provide a python script in my wiki. As soon as I uploaded it, I will paste the link into the comments and update the description in the Plugins Page

maocypher, 2020-11-04:
I uploaded a script into the repository. In the Wiki you can find a description how you can use this script till the feature is implemented and tested.
https://github.com/maocypher/Octoprint-Smart-Filament-Sensor/wiki

3DVERSEjn, 2020-11-10:
Im a little slow but could you elaborate on how to use the tester. I am having trouble getting the tester to work based on the description

3DVERSEjn, 2020-11-11:
I actually got it to work, my sensor is showing no signs of movement the wires are good but I might have fried the sensor

maocypher, 2020-12-28:
Hi, there is a pre-release now with integrated connection test and a visualization of the current values in the sidebar
https://github.com/maocypher/Octoprint-Smart-Filament-Sensor/archive/1.1.5.zip

maocypher, 2021-01-15:
Released
https://github.com/maocypher/Octoprint-Smart-Filament-Sensor/archive/1.1.5.3.zip

closed 2021-01-15 by maocypher