maocypher/Octoprint-Smart-Filament-Sensor #7: Octoprint not pausing the print when running out

dutchcolonial, 2020-11-09:
Hi, I hope it is okay that I post here, since I wasn't sure where else to post.
I have installed the BTT Smart Runout sensor on GPIO pin 24 and when running your test script, I see motion detected when I pull some filament through the sensor.
I configured the plugin and made sure the sensor is enable. I then started a print and had the test script running on the command prompt, just to monitor.
A little bit into the print I cut the filament and waited. The filament runs through the sensor and the test script does at that point only show "No motion detected". However, Octoprint never halts the print.
I have attached a screenshot of the settings and the captured output of the monitoring script. I have also attached a screenshot of the start GCODE I use.

[image]
[image]

[MonitoringResultstxt.txt](https://github.com/maocypher/Octoprint-Smart-Filament-Sensor/files/5513239/MonitoringResultstxt.txt)

maocypher, 2020-11-09:
Yes, it is totally fine asking here :)
That the test script is detecting, that no movement is detected anymore is good.
Distance detection is working a little different than the test script.
In logs octoprint.log I assume you can see that the sensor stopped detection.
Did you add M600 command to the pause commands of Octoprint as described in the instructions?

dutchcolonial, 2020-11-09:
I have to follow the code (in OctoPrint) for "After the print is paused"
. I had not added the M600 command, since I thought Octoprint would pause
the print and then I decide what to do. The M600 command is a firmware
command, right and would require me to use the LCD screen?
So I should  not use the below code anymore (I used it for a standard
filament runout detector) and just replace it with M600?
{% if pause_position.x is not none %}
; relative XYZE
G91
M83
; Lower extruder temp
M104 180

; retract filament, move Z slightly upwards
G1 Z+5 E-5 F4500

; absolute XYZE
M82
G90

; move to a safe rest position
 (use X0 and Y0 depending on your printer)
G1 X220 Y220
{% endif %}

maocypher, 2020-11-09:
You just need to add the command in Octopi -> settings -> serial connection -> firmware and protocol -> pause commands.

I am going to add more pause commands on the weekend, since M600 is causing endless loops with some LCD screens as I was informed

dutchcolonial, 2020-11-09:
Okay, currently the defaults seem to be M0, M1, M25. I think M0 and M1 would work for me and allow for putting GCODE in the "after pause" section of the octoprint GCODE section.

maocypher, 2020-12-25:
Fixed with https://github.com/maocypher/Octoprint-Smart-Filament-Sensor/tree/1.1.3

closed 2020-12-25 by maocypher