maocypher/Octoprint-Smart-Filament-Sensor #9: Only one detection per print

Oleynick07, 2020-11-10:
Here is my problem:
I print and filament jams/ends: plugin works fine. M600 is send correctly.
I continue print and filament jams/ends second time during the same print: plugin DOESN'T work. M600 is NOT send. I checked using that special testing script, that motion is NOT detected, but M600 is not send. When I send M600 manually, it works, so it's not printer problem. 
Do You have any idea if this is bug or configuration problem? If You want my help contact me at: pioole282@gmail.com. I'm Python programmer and I have a bit of free time, so I thing that I might be able to help :)

maocypher, 2020-11-10:
Hi :) could you look into the octoprint.log if the detection started again?
From my experiance it is too senstive and stopping to often, i.e. M600 is sent also a second, third etc. time.

Gianlucorso, 2020-12-20:
Hi, I have the same issue in my printer. In `octoprint.log` in `DEBUG` mode, I see the message `Sensor enabled: True` only once while, from reading your code and if I did not understand it wrong, it should print it at every `Events.PRINT_RESUMED` (I suppose every time the printer restarts printing after the filament change) and `Events.Z_CHANGE` (at every change of Z - new layer?).
Do you have any suggestion to solve this issue?
Thanks in advance,
Gianluca

Pavulon87, 2020-12-20:
Same problem for me. Changed __init__.py and looks like it is fixed:

> +++ __init__.py 2020-12-19 09:22:33.000000000 +0100
> @@ -170,6 +170,7 @@
>      # Reset the distance, if the remaining distance is smaller than the new value
>      def reset_distance (self, pPin):
>          self._logger.debug("Motion sensor detected movement")
> +        self.send_code = False
>          if(self.remaining_distance < self.motion_sensor_detection_distance):
>              self.remaining_distance = self.motion_sensor_detection_distance

[image]

maocypher, 2020-12-20:
I will check this in a few days. If you like do a pull request, so that I only need to do the merge ;)

Gianlucorso, 2020-12-21:
> Same problem for me. Changed **init**.py and looks like it is fixed:
>
>
> [image]

Hello,
I used the same solution proposed by @Pavulon87 quoted above, and it works fine.

maocypher, 2020-12-25:
https://github.com/maocypher/Octoprint-Smart-Filament-Sensor/commit/3d27668ca714229f39d4194909c5fff23b67abd5

PR #22 (2020-12-25): Fix Only one detection per print
Reset send_code with every movement
Fix #9 (dev branch).
closed 2020-12-25 by maocypher