maocypher/Octoprint-Smart-Filament-Sensor #10: Detection stopped, crash after disabling sensor

dertester, 2020-11-11:
Hi,

i had a problem today where i constantly got M600 spammed even if
filament was running fine. I could not test the sensor directly.

After if went into settings to disable the sensor it crashed my octoprint.

Hardware is a RPi 4 with
OctoPrint version : 1.4.2
OctoPi version    : 0.18.0

Here is my log:
[octoprint (7).log](https://github.com/maocypher/Octoprint-Smart-Filament-Sensor/files/5525148/octoprint.7.log)

maocypher, 2020-11-13:
Hi sorry for the late response, first of all your log looks strange, but I will analyse this.
What I see is that you are using a very small distance (5mm) which could cause that the sensor is reacting too sensitive.
You are also using M600 rewriter as plugin: in the next days I am going to provide a version with different pause events.

What is irritating me is that your print contiously changed between absolut and relative extrusion which might have caused this problem. Did you edit your GCode in OctoPrint afterwards? Because the GCode you provided is not changing that often...

dertester, 2020-11-13:
Hey thanks for the reply.

I thought 4mm was the stock value, for testing i changed it to 10 now. Seems to work fine for the last few prints.

Im not sure but i think the change in extrusion mode could be due to M600 rewrite to move the printhead away.
I have no display attached. I will test around with different settings and if you add pause events it would be really good for that case.

maocypher, 2020-11-13:
I have good experiance with 15 or 20mm. Sometimes 30mm was necessary, but you can try out for your sensor.
If the print pauses, you can change the value in the settings and then continue with the new value

maocypher, 2020-11-15:
Hi I released v1.1.3 with alternative pausing commands, that could solve your issues with M600

rechena, 2020-11-15:
Silly question, how do these work, do we still need to add them to the
Firmware commands?

maocypher, 2020-11-16:
There are no silly questions ;) Yes, if the pausing command is not preconfigured like M0 or M1, it is necessary to configure them. Otherwise OctoPrint is going the send the next GCode line. After the configuration a restart could be necessary.

dertester, 2020-11-23:
Hey,

thank you for enabling pause commands. I enabled emergency parser and use M0 now. With before and after gcode in octorprint i can move it away from my print.

I had problems prints often stopping. Thanks to your test code i found that some signal was "bleeding" into my connections. I am now using shielded cable and values look clean.

**Okay new question:**
Does chaning nozzle size affekt distance detection? 
I changed to 0.6 nozzle and i can see with your script WHILE printing (is running while printing a problem?) that its getting motion detected.
But often it just stops the print and im at 25mm already.

`2020-11-23 20:13:45,538 - octoprint.util.comm - INFO - Changing monitoring state from "Paused" to "Resuming"
2020-11-23 20:13:45,552 - octoprint.printer.standard.job - INFO - Print job resumed - origin: local, path: AC_ESP_Halter.gcode, owner: _api, user: _api
2020-11-23 20:13:45,563 - octoprint.plugins.smartfilamentsensor - INFO - Motion sensor started: Distance detection
2020-11-23 20:13:45,581 - octoprint.plugins.smartfilamentsensor - INFO - Motion sensor started: Distance detection
2020-11-23 20:13:45,586 - octoprint.util.comm - INFO - Changing monitoring state from "Resuming" to "Printing"
2020-11-23 20:17:52,828 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2020-11-23 20:23:21,804 - octoprint.plugins.smartfilamentsensor - INFO - Pause command: M0
2020-11-23 20:23:21,840 - octoprint.util.comm - INFO - Pausing print job due to command M0
2020-11-23 20:23:21,841 - octoprint.util.comm - INFO - Changing monitoring state from "Printing" to "Pausing"
2020-11-23 20:23:21,843 - octoprint.util.comm - INFO - Not sending M0 to printer, it's configured as a blocked command
2020-11-23 20:23:31,855 - octoprint.printer.standard.job - INFO - Print job paused - origin: local, path: AC_ESP_Halter.gcode, owner: _api, user: None
2020-11-23 20:23:31,858 - octoprint.plugins.smartfilamentsensor - INFO - PrintPaused: Pausing filament sensors.
2020-11-23 20:23:42,305 - octoprint.util.comm - INFO - Changing monitoring state from "Pausing" to "Paused"
2020-11-23 20:28:53,593 - octoprint.util.comm - INFO - Changing monitoring state from "Paused" to "Resuming"
2020-11-23 20:28:53,605 - octoprint.printer.standard.job - INFO - Print job resumed - origin: local, path: AC_ESP_Halter.gcode, owner: _api, user: dertester
2020-11-23 20:28:53,611 - octoprint.plugins.smartfilamentsensor - INFO - Motion sensor started: Distance detection
2020-11-23 20:28:53,631 - octoprint.plugins.smartfilamentsensor - INFO - Motion sensor started: Distance detection
2020-11-23 20:28:53,637 - octoprint.util.comm - INFO - Changing monitoring state from "Resuming" to "Printing"
2020-11-23 20:32:52,829 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2020-11-23 20:32:59,792 - octoprint.plugins.smartfilamentsensor - INFO - Pause command: M0
2020-11-23 20:32:59,828 - octoprint.util.comm - INFO - Pausing print job due to command M0
2020-11-23 20:32:59,829 - octoprint.util.comm - INFO - Changing monitoring state from "Printing" to "Pausing"
2020-11-23 20:32:59,831 - octoprint.util.comm - INFO - Not sending M0 to printer, it's configured as a blocked command
2020-11-23 20:33:09,841 - octoprint.printer.standard.job - INFO - Print job paused - origin: local, path: AC_ESP_Halter.gcode, owner: _api, user: None
2020-11-23 20:33:09,850 - octoprint.plugins.smartfilamentsensor - INFO - PrintPaused: Pausing filament sensors.
2020-11-23 20:33:20,292 - octoprint.util.comm - INFO - Changing monitoring state from "Pausing" to "Paused"`

I am not sure if im still loosing signals because of bad connections or 
if distance detection needs some tuning.

maocypher, 2020-11-23:
Thanks for the feedback. Yes, please run test script only before the print not while printing.

I am implementing connection test into the plugin, but I need some more time for it

maocypher, 2020-12-25:
Fixed with https://github.com/maocypher/Octoprint-Smart-Filament-Sensor/tree/1.1.3

closed 2020-12-25 by maocypher