maocypher/Octoprint-Smart-Filament-Sensor #12: Print keeps randomly stopping...

rechena, 2020-11-15:
Not sure if I have anything wrong byt after 2 hours printing my print just got stoped. 
Looking at the test to see if the print is running and and the logs theres no apparent reason for this to happen.. 

Any ideas?

```
No motion detected
Motion detected at 1605455890.7617376
Moving
Moving
Moving
Motion detected at 1605455891.5431175
Moving
Moving
Moving
Motion detected at 1605455892.123839
Moving
Moving
Moving
Motion detected at 1605455892.9361026
Moving
Motion detected at 1605455893.309376
Moving
Moving
Motion detected at 1605455893.5848026
Moving
Motion detected at 1605455893.9593928
Moving
Motion detected at 1605455894.2271972
Moving
Moving
Motion detected at 1605455894.5909858
Moving

```
2020-11-15 15:56:33,228 - octoprint.plugins.smartfilamentsensor - INFO - PrintPaused: Pausing filament sensors.
2020-11-15 15:57:48,041 - octoprint.util.comm - INFO - Changing monitoring state from "Pausing" to "Paused"
2020-11-15 15:57:50,018 - octoprint.util.comm - INFO - Changing monitoring state from "Paused" to "Resuming"
2020-11-15 15:57:50,029 - octoprint.printer.standard.job - INFO - Print job resumed - origin: local, path: Terminator/T800_Base_Supported_v5.4.5.gcode, owner: _admin, user: _admin
2020-11-15 15:57:50,035 - octoprint.plugins.smartfilamentsensor - INFO - Motion sensor started: Distance detection
2020-11-15 15:57:50,042 - octoprint.util.comm - INFO - Changing monitoring state from "Resuming" to "Printing"
2020-11-15 15:57:54,625 - octoprint.util.comm - INFO - Pausing print job due to command M600
2020-11-15 15:57:54,625 - octoprint.util.comm - INFO - Changing monitoring state from "Printing" to "Pausing"
2020-11-15 15:58:04,645 - octoprint.printer.standard.job - INFO - Print job paused - origin: local, path: Terminator/T800_Base_Supported_v5.4.5.gcode, owner: _admin, user: None
2020-11-15 15:58:04,647 - octoprint.plugins.smartfilamentsensor - INFO - PrintPaused: Pausing filament sensors.
```

Thanks for the help

rechena, 2020-11-15:
These are the settings I have

[image]
[image]

maocypher, 2020-11-15:
Hi, if it happens suddenly, please increase the detection distance first. Sometimes the sensors detect movement later, than they should.
Good values to go are 25 or 30

rechena, 2020-11-15:
Will give it a try.. been waiting to get the sensor setup to print this 36 hours print 😂

rechena, 2020-11-15:
As soon I've enabled it and changed the settings to 30mm the  same thing happened again :( 

```
2020-11-15 16:23:53,185 - octoprint.plugins.smartfilamentsensor - INFO - Motion sensor started: Distance detection
2020-11-15 16:23:53,193 - octoprint.util.comm - INFO - Changing monitoring state from "Resuming" to "Printing"
2020-11-15 16:24:04,796 - octoprint.util.comm - INFO - Pausing print job due to command M600
2020-11-15 16:24:04,797 - octoprint.util.comm - INFO - Changing monitoring state from "Printing" to "Pausing"
2020-11-15 16:24:14,815 - octoprint.printer.standard.job - INFO - Print job paused - origin: local, path: Terminator/T800_Base_Supported_v5.4.5.gcode, owner: _admin, user: None
2020-11-15 16:24:14,818 - octoprint.plugins.smartfilamentsensor - INFO - PrintPaused: Pausing filament sens
```

But on the test script I still see:

```
Motion detected at 1605457478.587688
Moving
Moving
Moving
Moving
Motion detected at 1605457479.7159672
```

:(

rechena, 2020-11-15:
Wondering if its somehting with my PI.. cause when I make changes on Octopi interface the test script only shows:

```
No motion detected
No motion detected
No motion detected
No motion detected
No motion detected
No motion detected
No motion detected
No motion detected
No motion detected
```
I need to CTRL+C and start the python script again to see something.... wierd...

maocypher, 2020-11-15:
Did you check your wiring?

The test script always starts with 2 secs motion detected. If your print is currently running you should increase the time to e.g. 30secs.
The easiest way to test is moving the filament manually by hand with 2sec detection.

maocypher, 2020-11-15:
I ran a 24h print with 30mm without false interuption in the mid of the print job my filament ran out. If there is a lot of retraction it could be smaller I assume.

My print had mostly long lines and less retraction
https://www.thingiverse.com/thing:4636413

rechena, 2020-11-15:
> Did you check your wiring?
> 
> The test script always starts with 2 secs motion detected. If your print is currently running you should increase the time to e.g. 30secs.
> The easiest way to test is moving the filament manually by hand with 2sec detection.

The wiring was my first suspicious, but checked and everything seems to be fine. What I'm find weird is the plug-in stopping the print but the test python is reporting "movement". Could it be because the python os running that affects the print?

rechena, 2020-11-15:
> I ran a 24h print with 30mm without false interuption in the mid of the print job my filament ran out. If there is a lot of retraction it could be smaller I assume.
> 
> My print had mostly long lines and less retraction
> https://www.thingiverse.com/thing:4636413

I'm currently printing the base of the T800 terminator head so loads of long lines and almost 0 retraction :( 
Before this I printed the Ultimaker small robot without issues...

maocypher, 2020-11-15:
You could do an experiment and set it even higher. Perhaps 40 or 50. I had the feeling, that if the printed lines are too long and too slow that sensor skips detections

rechena, 2020-11-15:
Ok I think the issue might be related to running the python script. 
Even with the 40mm the issue was still hapening. I would leave the ptyhon loop running and make the changes on octoprint, immediately the script would start outputting 

```
No motion detected
No motion detected
No motion detected
No motion detected
No motion detected
No motion detected
No motion detected
No motion detected
No motion detected
No motion detected
```

So I've made the change, Pause and Unpaused the print and didn't start up the test script again, and seems to be working just fine now.

Two things that are curious, one, does it make some sort of "lock" on the GPIO pin when I save on Octoprint? Or the other way around the "lock" was being done by running the the

```
$  python3 filament_motion_sensor_connection_check.py

```

Can you test it?

Thanks

maocypher, 2020-11-15:
I could test is, but currently I am implementing the connection test into the plugin and a visualization, so that you can see if movement is detected.

I.e. in the near future it is not necessary anymore.

If you set the log level to debug for the plugin you can follow the remaining distance being logged.

If you start the script and then start/resume the print it might happen that all callbacks for the events I subscribe are removed. I am not sure how thread save it is implemented in the GPIO library

rechena, 2020-11-15:
> If you set the log level to debug for the plugin you can follow the remaining distance being logged.
Is this something we can do now?

Looking forward to the new implementations to the plugin. Btw great work so far, thank you for doing this the plugin is amazing :) 

Yeah so far the print has not stopped, #fingerscrossed. Wonder if it will stop for a real reason now?

rechena, 2020-11-16:
I was thinking on this and I was wondering if we could somehow report on the octoprint ui the status of the plug-in? Same the test does. Would be cool to just see "moving"/"stopped". Not sure if possible, or even a thing, maybe just my ocd 😂

maocypher, 2020-11-16:
This is what I am currently implementing ;)
You could sneek into it, if you install the version from branch sidebar-gui, but it is still under development and has some bugs, because I had to do a complete code refactoring.

I also thought about these features:
- some intelligence that calculates a recommended distance
- maybe a line graph that displays the current distance, length of the line and maybe some other important information that could help understanding how to find the best configuration

rechena, 2020-11-16:
Will defo give it a try when this 38 hours print end..:)
Those ideas that you have are awesome, what are you thinking for the
"calculation" Machine learning?

maocypher, 2020-11-16:
I thought about it, but before I must dig into ML algorithms ^^"
If you have experiance it would be great if you could share it

maocypher, 2020-11-16:
For the beginning I thought about simple avaerage calculation that is permanently saved and increased

rechena, 2020-11-16:
> I thought about it, but before I must dig into ML algorithms ^^"
> If you have experiance it would be great if you could share it

My ML is quite limited, but I'm reaching out to a friend which is is main area :) I'll ask him to comment here.

rechena, 2020-11-16:
Ok he's willing to help, do you have data that we can use in the model training? Or anywhere were we can get that data?

maocypher, 2020-11-16:
The data exist as soon as you switch log level to debug. But not in a nice format.

If you like you could join my discord channel for the development of this topic: https://discord.gg/YQsXrSwrV7

closed 2020-11-20 by maocypher