maocypher/Octoprint-Smart-Filament-Sensor #13: Test/debug sensor.

strangecalling, 2020-11-17:
I have set up the plug-in using a BTT SFS connected to pin 8 GPIO14 but the print does pause when the filament is stationary.

Is there a log file to view to show switch is functioning? Someway to check setup?

maocypher, 2020-11-17:
Hi, you can find the logfiles in octoprint.log: https://community.octoprint.org/t/where-can-i-find-octoprints-and-octopis-log-files/299
If you want to change the loglevel you must add octoprint.plugins.smartfilamentsensor

But what do you mean with stationary? Not moving or stuck?

Your GPIO TXD has special funtionality https://www.raspberrypi.org/documentation/configuration/uart.md

strangecalling, 2020-11-17:
I will check the logs and change the log level if needed.

By Stationary I mean filament not moving by disabling the extruder drive arm.

Do GPIO with special function not get assigned as inputs like normal GPIO pins? If so I will wired to a standard pin.

maocypher, 2020-11-18:
It might be necessary to disable this funtionality in the pi config first, so the signals are not mixed up.

If still GCodes were send by OctoPrint during the stationary, then it is correct that the print can pause.

It is meassured by G0, G1 in future also G2, G3 commands. Maybe your configuration was too small if everything else was working fine

maocypher, 2020-12-25:
Hi is there any news about this issue? If not it will be closed in a few days

maocypher, 2020-12-28:
Hi, there is a pre-release now with integrated connection test and a visualization of the current values in the sidebar
https://github.com/maocypher/Octoprint-Smart-Filament-Sensor/archive/1.1.5.zip

maocypher, 2021-01-15:
Integrated test is released
https://github.com/maocypher/Octoprint-Smart-Filament-Sensor/archive/1.1.5.3.zip

closed 2021-01-15 by maocypher