maocypher/Octoprint-Smart-Filament-Sensor #14: Request

sushibagel, 2020-11-20:
Ignore ironing - I ran a print today that had ironing on the top layer. The print paused because of the ironing. 

Is there anything that can be done to ignore a lack of filament movement during ironing?

maocypher, 2020-11-20:
Hi, can you provide me the ironing GCode lines? Then I am going to analyse this

sushibagel, 2020-11-20:
I'm not sure what the ironing specific code would be unfortunately but here is the last layer 

`;LAYER:33
;TYPE:WALL-INNER
;MESH:Cable Management Cover.stl
G1 F300 Z6.9
G1 F3000 E3835.7087
G1 F1800 X108.373 Y123.204 E3835.71755
G1 X108.645 Y122.859 E3835.73216
G1 X108.966 Y122.283 E3835.75409
G1 X109.281 Y121.259 E3835.78972
G1 X109.295 Y121.038 E3835.79709
G1 X109.319 Y119.968 E3835.83269
G1 X108.979 Y118.745 E3835.87491
G1 X108.516 Y117.939 E3835.90582
G1 X108.352 Y117.755 E3835.91402
G1 X107.43 Y116.872 E3835.95648
G1 X106.266 Y116.361 E3835.99876
G1 X105.998 Y116.265 E3836.00823
G1 X104.918 Y116.155 E3836.04434
G1 X103.867 Y116.29 E3836.07958
G1 X102.519 Y116.929 E3836.1292
G1 X102.302 Y117.103 E3836.13845
G1 X101.736 Y117.62 E3836.16394
G1 X101.094 Y118.581 E3836.20238
G1 X101 Y118.828 E3836.21117
G1 X100.664 Y120.036 E3836.25288
G1 X100.672 Y120.321 E3836.26236
G1 X100.769 Y121.568 E3836.30396
G1 X101.316 Y122.809 E3836.34907
G1 X101.55 Y123.127 E3836.3622
G1 X101.659 Y123.301 E3836.36903
G1 X102.548 Y124.103 E3836.40885
G1 X102.743 Y124.203 E3836.41614
G1 X102.786 Y124.225 E3836.41775
G1 X103.639 Y124.6 E3836.44874
G1 X102.786 Y124.6 E3836.46172
G1 X102.737 Y124.6 E3836.4633
G1 X83.537 Y124.6 E3837.1019
G1 X83.35 Y124.58 E3837.10815
G1 X83.094 Y124.57 E3837.11667
G1 X82.562 Y124.463 E3837.13472
G1 X82.448 Y124.427 E3837.1387
G1 X81.793 Y124.157 E3837.16226
G1 X81.466 Y123.968 E3837.17482
G1 X80.999 Y123.553 E3837.1956
G1 X80.777 Y123.338 E3837.20588
G1 X80.572 Y123.089 E3837.21661
G1 X80.45 Y122.848 E3837.22559
G1 X80.313 Y122.632 E3837.2341
G1 X80.029 Y121.945 E3837.25882
G1 X80.019 Y121.904 E3837.26023
G1 X79.915 Y121.117 E3837.28663
G1 X79.901 Y120.974 E3837.29141
G1 X79.901 Y89.021 E3838.35417
G1 X79.928 Y88.638 E3838.36694
G1 X79.943 Y88.498 E3838.37162
G1 X80.222 Y87.528 E3838.4052
G1 X80.288 Y87.403 E3838.4099
G1 X80.698 Y86.757 E3838.43534
G1 X80.845 Y86.595 E3838.44262
G1 X81.074 Y86.364 E3838.45344
G1 X81.792 Y85.84 E3838.483
G1 X81.846 Y85.81 E3838.48506
G1 X82.826 Y85.501 E3838.51923
G1 X83.103 Y85.428 E3838.52876
G1 X83.446 Y85.403 E3838.5402
G1 X83.515 Y85.401 E3838.5425
G1 X133.753 Y85.4 E3840.21342
G1 X133.754 Y85.4 E3840.21345
[... 2625 lines cut ...]
G1 X140.369 Y98.932 E4084.04491
G1 X134.787 Y93.351 E4084.06198
G1 X134.611 Y93.316 E4084.06237
G1 X140.369 Y99.074 E4084.07997
G1 X140.369 Y99.215 E4084.08028
G1 X134.418 Y93.264 E4084.09847
G1 X134.09 Y93.077 E4084.09929
G1 X140.369 Y99.356 E4084.11848
G1 X140.369 Y99.498 E4084.11879
G1 X133.761 Y92.89 E4084.13899
;TIME_ELAPSED:7706.841242
G1 F3000 E4083.13899
M140 S0
M107
G91 ;Relative positioning
G1 E-2 F2700 ;Retract a bit
G1 E-2 Z0.2 F2400 ;Retract and raise Z
G1 X5 Y5 F3000 ;Wipe out
G1 Z10 ;Raise Z more
G90 ;Absolute positionning

G1 X0 Y210 ;Present print
M106 S0 ;Turn-off fan
M104 S0 ;Turn-off hotend
M140 S0 ;Turn-off bed`

maocypher, 2020-11-20:
Yep, it should be the last layer.
I try to check. But it might take some time, because I received new hardware for my printer

sushibagel, 2020-11-20:
> Yep, it should be the last layer.
> I try to check. But it might take some time, because I received new hardware for my printer

No worries, it's easy enough to just turn off the plugin temporarily if I'm using ironing. I just thought it would be a cool enhancement to the plugin if there was a way to work around it

maocypher, 2020-11-20:
I absolutly agree. I let you know, as soon as I have news

MADAL1EN, 2021-03-02:
Any plans to fix this issue?

maocypher, 2021-03-06:
Yes, there is, but till summer it looks like I have to much work, to continue... so, there will be development break till summer