maocypher/Octoprint-Smart-Filament-Sensor #15: Pausing does not use gcode pause script

Diilandh, 2020-12-06:
When i manually pause a print or send my pause command through the terminal it pauses my print and moves the printhead to the side but when the filamentsensor plugin pauses it just stops on top of my print. 
I chose the M1 command as a pause command because the M600 does not work when i try this in the terminal but the M1 worked perfect but when the sensor is triggered and pauses my print it just stops on the spot. Is there any setting i overlooked? Before i had a bigtreetech filament sensor i just had a switch and when that one paused my print it moved the head aside.
M1 is in my pause commands, i set my gcode scripts for pausing and resuming and they work when i manually test them. I tried to find a solution online but i did not find anything the problem i am having.

Thanks in advance 

Sincerely

maocypher, 2020-12-07:
Good evening, currently I don't have a propper explanation for this behaviour.

Could you provide me your setup (Octoprint version, configuration of the plugin and the firmware & protocol information)?

Diilandh, 2020-12-07:
Certainly, i am currently running octoprint version 1.5.1, the smart filament sensor plugin is version 1.1.3. In the firmware and protocol i have the following under pausing commands M0, M1, M25, M600
I see that M0 and M1 also are under the blocked commands and i cant delete those they just return. (I am new to this hobby i still have a lot to learn). While printing a test file a tested all of the pause commands available in the terminal and only m0 and m1 worked so i picked m0. I was printing a somewhat larger than my usual prints and it paused twice but thats because i have to edit my detection distance and the first time it just stopped on top of my print but the second time it did the pause correctly and moved the printhead to the side. My plugin configuration is board mode, distance detection, M0-stop ( i also tried m1 but had the same result) and detection distance was 15mm i now updated it to 30mm

Thanks for your response 

Sincerely

maocypher, 2020-12-08:
I run a test print later today and check if I run into the same issues. Maybe something changed with the new octoprint version

maocypher, 2020-12-08:
I ran a test and it worked fine for me. So, it would be necessary to take a closer look on your configuration. Maybe you upload one or two screenshots?

maocypher, 2020-12-25:
Hi :) is there any news about this? If not then I will close this issue in a few days

maocypher, 2020-12-28:
Hi, there is a pre-release now with integrated connection test and a visualization of the current values in the sidebar
https://github.com/maocypher/Octoprint-Smart-Filament-Sensor/archive/1.1.5.zip

Diilandh, 2020-12-28:
Hello

I am sorry for the late response. I didn´t have time to test this further I had some large prints i needed to make and i removed the sensor for now so it didn't interfered with my prints, in a few days everything is finished and i will do more tests with the sensor. 

Thanks for your response

Sincerely

maocypher, 2020-12-28:
Okay, no problem :) let me know if there are any news

Diilandh, 2020-12-28:
Hello

I had to switch my filament and didn't had the patience to wait longer so i did some tests with the new version of your software. First of all the new connection test and sidebar information feature is really a must have. That is an amazing feature :D. The connection test works great and my sensor works like it should, when i did a test print i can see that the distance keeps counting down until it hits zero and than it pauses my print and it paused it like it should, it moved the head away so the problem seems to be gone. I however faced a new problem when i move the filament it does not register it and it doesn't reset the countdown it just keeps counting down until the pause, but when I do the connection test it worked perfect it says my filament moves when i move it and it says stop when i dont move it but during a print it does not register it. When i switched from distance detection to timeout detection it works like a charm. It updates the time when I move the filament. I noticed it also updates the time when i retract the filament so i have a suggestion. Could it be possible to only reset the countdown when the filament is moving forward? I had a problem in the past that my filament was stuck on the spool and my extruder just kept skipping when it was supposed to extrude but it retracted just fine so even if I had this sensor installed back then it wouldn't have helped because the retraction would keep resetting the countdown. 

Thanks in advance

Sincerely

maocypher, 2020-12-28:
What configuration did you use for distance detection?

Diilandh, 2020-12-28:
Hello

The distance is set to 15 but i noticed that the countdown starts from 22 and in the sidebar the filament movement is either constant yes or no it does not change.

Sincerely

maocypher, 2020-12-28:
That is correct. There is a buffer value for the first movement after pause, because more filament is extruded.

15mm is a bit small. Try to increase in 5mm steps, to find out your configuration.

Filament is moving can only show the last state. The measurement is done every x mm that passed the sensor. So it is hard to find out between the measurement points. The signal is changing from 1-0-1-... with every turn of the wheel inside the sensor