maocypher/Octoprint-Smart-Filament-Sensor #16: Dual extruder system question

Drzet, 2020-12-15:
I have a single sensor attached to a E2 extruder. Another extruder doesn't have any sensors on it.
So far I had one instance of filament jamming detection, but that was a single extruder print. 
During a dual print I had no trigger events even though there is no filament movement while E1 is extruding
Question:
How is this plugin supposed to behave during dual extrusion print?
Is it possible to have 2 filament sensors ?

maocypher, 2020-12-16:
Hi, currently this feature is under development.
If you can provide me a gcode snippet I can check if there is something different.
A volunteer checked a version for 4 extruders that worked so far.
But the detection on the one extruder should work anyway

Drzet, 2020-12-16:
Sorry, I am not sure what "g-code snippet" means.
How does this plugin know which extruder it is monitoring?

maocypher, 2020-12-16:
With GCode snippet I mean a few lines from your gcode.
I assume there is a tool change somewhere.
This is currently not supported, but under development.

G1 and G0 commands don't contain an identifier which extruder is used. I.e. if the tool is changed the passed distance is still messures, but no movement is detected.

I can keep this in mind to support less sensors then extruders.

Drzet, 2020-12-16:
Thanks for the explanation. I've checked my dual extrusion gcodes, it's simple T0 and T1 commands.
From what you said, it sounds like the sensor _should_ trigger while the tool with NO sensor attached is printing, right ? I am surprised it does not.

closed 2020-12-25 by Drzet