maocypher/Octoprint-Smart-Filament-Sensor #17: Sensor detected but printer pauses anyway

Haku73, 2020-12-18:
I tried with the script and put in BCM mode GPIO 0. Sensor works regularly with the detect/no detect of the movements. I put the same data inside the plug-in (BCM mode, GPIO 0) but after a few moments the printer (a CR10 S4) pauses as if it didn't detect it. Tried different lengths of sensitivity. What could I try to do? Thanks

maocypher, 2020-12-18:
Hi, is the test script running in parallel? This leads to the problem that the events are blocked.

Haku73, 2020-12-18:
yes, I close the script then and try again.

Haku73, 2020-12-18:
Hi, unfortunately it continues to pause and when I restart it from the dock (Octoprint does not reveal the pause) from that moment the plugin is bypassed and to "reactivate" I have to pause and resume the print from Octoprint but with the same problem. Thanks

maocypher, 2020-12-19:
What configuration of the sensor are you running? And is the print speed rather slow or fast?
The bigger the print and the slower the speed, the higher the diatance detection must be, some people figured out.
You can change the configured distance while the print paused

Haku73, 2020-12-20:
BCM mode GPIO 0. Slow. 
The strange thing is that with the debug script it works (at least the timeout based on time) but the plugin doesn't. I use them alternatively not together with plug-in. For example, if I put max_not_moving_time = 10 in the script, it doesn't reveal any unwanted interruptions. If I put max_not_moving_time = 0 I see every single movement. Perfect.

If I put the same variables in the plugin nothing happens. He never intervenes.

For the distance detection instead happens the opposite, The printer always enters the pause before or after depending on the distance variable but it looks like it doesn't take sensor data in cosideration

Bye & Thanks.

maocypher, 2020-12-20:
If you are printing very slow, then the distance need to configure could be greater then 50mm, because the sensor is not as sensitive to really recognize every 7mm. In the firmware it must be solved differently. But from my experiance many signals are skipped or the filament is slipping in the sensors.
Maybe it has something to do with running it with 3.3V, but 5V is too high for GPIO input

Haku73, 2020-12-20:
I tried at different speeds but unfortunately the behavior doesn't change. It would be useful inside the plugin a detector that tells if the sensor works or not so as to understand if the port is read regularly and adjust the sensitivity but I guess it is under development.

maocypher, 2020-12-20:
Yes, the view to see changes is under development.
You could take a look into octoprint.log and search for errors

maocypher, 2020-12-28:
Hi, there is a pre-release now with integrated connection test and a visualization of the current values in the sidebar
https://github.com/maocypher/Octoprint-Smart-Filament-Sensor/archive/1.1.5.zip

dertester, 2021-01-09:
Hey i tried the pre release but its not working for me.

Its just stuck at start test.

AttributeError: 'SmartFilamentSensor' object has no attribute 'motion_sensor_thread'

[octoprint (4).log](https://github.com/maocypher/Octoprint-Smart-Filament-Sensor/files/5791777/octoprint.4.log)

maocypher, 2021-01-09:
Okay, I take a closer look at it

dertester, 2021-01-09:
Hey,
thank you for your effort making this work.

I tested some more.
In __init__.py  #self.remaining_distance = self.motion_sensor_detection_distance was commented. (Pre release branch

I uncommented it and now the test is working and showing me movement.
But while printing its not resetting the value
Filament is moving: No
Remaining distance: 15.988779999999963 and slowly ticking down.

But i really like the sidebar. I think it will come in handy to calibrate the lenght.

maocypher, 2021-01-10:
There is a pre-release from Friday with a fix for the resetting

maocypher, 2021-01-11:
I checked the connection test in both releases and it works for me. Please check if you have the same error on the latest one.
https://github.com/maocypher/Octoprint-Smart-Filament-Sensor/archive/1.1.5.1.zip

I installed from the URL. If the error still occurs please try to delete your plugin data and configure the plugin again

dertester, 2021-01-11:
Hey, thx.

Still the same issue.
```
File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_smart_filament_sensor/__init__.py", line 146, in start_connection_test
    if(self.motion_sensor_thread == None):
AttributeError: 'SmartFilamentSensor' object has no attribute 'motion_sensor_thread'
```
[octoprint (12).log](https://github.com/maocypher/Octoprint-Smart-Filament-Sensor/files/5797281/octoprint.12.log)

I did delete the plugin data.

1.1.4 detection while printing works
1.1.5.1 Error above, click on test wont do anything
1.1.5.1 with uncommented line 25 in __init__.py the test works but printing does not reset counter => always pauses

maocypher, 2021-01-11:
Is it during the connection test or while printing?

The strange thing is, I did a fresh installation. I try again tomorrow, maybe something during the merge went wrong. I tried rebase instead of merge for the first time

dertester, 2021-01-11:
I did not try to print without commented line. 

When I go into settings and click start test it won't change the button and just won't do anything.

I changed my last comment. It's working with line uncomment while printing.

Remain resets but status is always filament not moving.

Hope that helps.

maocypher, 2021-01-11:
Which browser do you use?

dertester, 2021-01-11:
Chrome win64

I haven't tried to clear cache. It's just something I noticed.

maocypher, 2021-01-11:
It would be good, if you give it a last try. Then I have better idea what I am looking for.

I am going to uninstall the plugin an reinstall it and test it again

dertester, 2021-01-11:
I tried with Chrome cache cleared, Edge and Chrome Android and 
even if its working fine it shows "Filament is moving: No".

For your zip i always have to uncomment line 25. I made sure its deinstalled with restart and that the old pluginfolder was deleted in octoprint before i installed the zip.

Sensor enabled: Yes
Detection mode: Distance Detection
Filament is moving: No
Remaining distance: 13.23848000000001
Connection Test is running: Stopped

Another thing that might be confusing/cosmetics: There seems to be a 5mm buffer at start/pause that is shown in the sidepanel. 
That was the reason why i at first thought it wasnt working because these extra 5mm did not add/reset when moving filament.

maocypher, 2021-01-11:
If you press close button the connection test should stop.
You should stay on the view with start and stop buttons

I couldn't wait and checked it again on my mobile. Deleted everything of the plugin installed from URL and it works as designed

dertester, 2021-01-11:
Here is everything in a video:
https://youtu.be/W1r8DtEt39I

You can see the problem and after i unccoment how it works.
I did not talk but i guess you can see most of it.

ABL takes some time but at the end you see that i have to dis/enable the plugine once (could also be a too little setting) and then it works without showing filament moving. Now that i think about it maybe that refers only to test mode?

maocypher, 2021-01-11:
I definately need to take a closer look at it, because I did the very same and it worked...

As I can see you used the zip file from the link I provided. I let you know as soon as I found something out.

Erst nach 1min ist mir aufgefallen, dass die GUI in deutsch ist

maocypher, 2021-01-11:
I found out what causes the issue, after changing the pin and to BCM a restart is needed.
I try to fix it the next days ;)

maocypher, 2021-01-12:
I hopefully fixed it, so that no restart of Octoprint is needed
https://github.com/maocypher/Octoprint-Smart-Filament-Sensor/archive/1.1.5.2.zip

dertester, 2021-01-12:
Hey,

connection test works fine now without the need to change code.

But for my first print i still had to disable and enable sensor in plugin setttings once after the first few mm. Did not restart octoprint or the print.
I attach my log
[octoprint (15).log](https://github.com/maocypher/Octoprint-Smart-Filament-Sensor/files/5802051/octoprint.15.log)

What i did:
Uninstall 1151
Install 1152
Restart octoprint
setup plugin, save settings
connection test => works, stopped
start print => pause after first mm
Plugin settings: Disable, save, enable, save
Resume print => works!

Another thing i saw:
Had a print over night and this morning after it finished remaining filament was in the range of 3k or maybe 30k. Will have a look for this after the next print.

maocypher, 2021-01-12:
The configures distance depends on print size and speed. Also values from 60mm or even 100mm might be necessary.

That the value in the end is very high could be caused by the change from absolut to relative extrusion at the end of the print

maocypher, 2021-01-15:
Released
https://github.com/maocypher/Octoprint-Smart-Filament-Sensor/archive/1.1.5.3.zip

jamie45678, 2021-02-16:
It looks like I'm having a similar issue. I made a post on it before I read this one. I'm on 1.1.5.3.