maocypher/Octoprint-Smart-Filament-Sensor #18: Failed to install

urishani, 2020-12-20:
Trying to install failed. Tried with bundled plugin manager stalls and does not finish installation. Trying with the URL for the zip file acts the same - no progress and I had to refresh octoprint on my browser.
Download the zip file and trying gives me an "unknown" error with no additional infomation.

I have rebooted octoprint and the same results.
Running octopi.
Version is the latest: 
OCTOPI: Version 0.17.0, running on Raspberry Pi 2 Model B Rev 1.1
OCTOPRINT: Version 1.5.0
Just ordered from bigtreetech via ebay. Hope I can use it with thie plugin.
Thanks for any suggestions.

maocypher, 2020-12-20:
Good morning :) could you also provide a list of your installed plugins?

urishani, 2020-12-20:
[image]
[image]

maocypher, 2020-12-20:
You could try to disable the first two plugins and try again to install.

When I had the Spaghetti Detective Enabled I had trouble with restarting Octoprint once

maocypher, 2021-01-08:
Hi, did you try again to install the plugin?