maocypher/Octoprint-Smart-Filament-Sensor #19: ERROR - Could not save settings

pheuts, 2020-12-20:
Im not getting the plugin to work. In Octoprint's log I find the following after changing a setting for the plugin:

2020-12-20 16:00:02,858 - octoprint.plugins.smartfilamentsensor - INFO - Using BCM Mode
2020-12-20 16:00:02,859 - octoprint.server.api.settings - ERROR - Could not save settings for plugin Smart Filament Sensor (1.1.3)
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/api/settings.py", line 1034, in _saveSettings
    plugin.on_settings_save(data["plugins"][plugin_id])
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/util/__init__.py", line 1890, in wrapper
    return f(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_smart_filament_sensor/__init__.py", line 116, in on_settings_save
    self._setup_sensor()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_smart_filament_sensor/__init__.py", line 77, in _setup_sensor
    GPIO.setmode(GPIO.BCM)
ValueError: A different mode has already been set!

Not being able to save settings obviously would prevent the plugin from working correctly.
Am I doing something wrong? Forgetting a setting somewhere? Or is this a bug?

maocypher, 2020-12-20:
Maybe Python 2 is behaving different from Python 3. I can take a closer look on this in a few days.

Is a print running while you tried to change the setting?

pheuts, 2020-12-20:
There was no print running while changing the setting. 

I wil see if i can update Python, and if this solves the problem. I did not consider the Phyton version as a cause in this. Thanks for the suggestion.

pheuts, 2020-12-20:
@maocypher You were absolutely wright about the Phyton version being the cause of this problem. I updated to Phyton 3, and immediately everything started working as it should. Thank you for your help!

closed 2020-12-20 by pheuts

maocypher, 2020-12-20:
Thanks for the feedback. Anyway I should fix it for Python 2 ;)