maocypher/Octoprint-Smart-Filament-Sensor #20: Custom pause command

Pavulon87, 2020-12-20:
It would be nice if there was an option for custom pause command.

I'm using `M118 action:runout_detected` just to inform about filament problems. Sometimes it's just a problem with the sensor and the M600 isn't needed.

I changed file smartilamentsensor_settings.jinja2 but maybe someone would also use it.

maocypher, 2020-12-21:
Hi, I could add it, but it is on own risk then (only for advanced users)

ryancasler, 2021-01-26:
It would actually be a lot easier if it simply had the option to execute the OctoPrint Pause function rather than sending the command to the printer.  For example, my printer, the Tronxy XY-2 Pro,  doesn't really know that it's printing when printing from Octoprint.  It's simply executing GCODE command.  So, sending any of the pause GCODE commands to the printer won't really do anything.    What I really need to have happen is OctoPrint stop sending commands to the printer.

maocypher, 2021-01-26:
Did you also try e.g. M1 command?
If I don't add the commands to OctoPrint configuration to be interpreted as pausing command also my printer just continues printing.

Like this
[image]

ryancasler, 2021-02-12:
No, I mean invoking a pause within Octoprint.  To have Octoprint execute the built in pause function, the same as if you were to hit pause on the UI.  That is not just sending commands to the printer.  There are other functions executed as well, such as the custom pause script.

maocypher, 2021-02-12:
I already received a pull request for the default octoprint pause command. I still need to test it, before I release it. But it is in the pipe

ryancasler, 2021-02-19:
> I already received a pull request for the default octoprint pause command. I still need to test it, before I release it. But it is in the pipe

I tried your branch of the repo and it didn't work.  After a pause the head just sat there and did not execute the Octoprint pause script.  I know the Pause Script works when I press the pause button or put it into gcode manually.  Is it possible that the code injection is happening after Octoprint has already tried to detect the @pause in the gcode?

Royrdan, 2022-01-15:
I added a pull request for the @pause and it works perfectly for me.