maocypher/Octoprint-Smart-Filament-Sensor #24: False Detection Issues

Chrosus, 2020-12-29:
I was receiving false detections mid print using 1.1.4, so I installed the 1.1.5 pre-release build that was provided in other issues here. I ensured that the pin number was working using the new test feature. When running the plugin on the Distance Detection mode, I watched the "distance remaining" value tick down to zero as the print continued, triggering a runout detection. I watched the sensor the whole time to ensure the filament was still moving through, which it was. I also tried various values of distance ranging from 15 up to 50, and had the same result (print speed was between 50 and 70mm/s). 
I switched to Timeout Detection mode and had better luck; it was set to the default 45 seconds. However now I have an issue where it triggers runout detection before my printer even prints the first pre-print test lines. Curious if that has something to do with my auto bed leveling procedure taking too long, or perhaps because I perform a G1 command to get to the start position before reaching temperature. I bolded the line I'm referring to.

Here is my Start G-code:
; Ender 3 Custom Start G-code
M140 S{material_bed_temperature_layer_0} ; Set Heat Bed temperature
M190 S{material_bed_temperature_layer_0} ; Wait for Heat Bed temperature
M104 S160; start warming extruder to 160
G28 ; Home all axes
G29 ; Auto bed-level (BL-Touch)
G92 E0 ; Reset Extruder
M104 S{material_print_temperature_layer_0} ; Set Extruder temperature
**G1 X0.1 Y20 Z10.0 F5000.0 ; Move to pre-heat position**
M109 S{material_print_temperature_layer_0} ; Wait for Extruder temperature
G1 X0.1 Y20 Z0.3 F5000.0 ; Move to start position
G1 X0.1 Y200.0 Z0.3 F1500.0 E15 ; Draw the first line
G1 X0.4 Y200.0 Z0.3 F5000.0 ; Move to side a little
G1 X0.4 Y20 Z0.3 F1500.0 E30 ; Draw the second line
G92 E0 ; Reset Extruder
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
; End of custom start GCode

maocypher, 2020-12-29:
Hi :) sometimes a distance higher than 50mm is necessary. I also used 60mm others 100mm.

For the other issue: did you try restarting Octoprint?

Chrosus, 2020-12-29:
I suspect that because my PIDs aren't super well tuned, it took longer than 45 seconds to reach temp, after it did the G1 movement. I moved the G1 command to after my temp is reached and that seemed to have fixed the issue using the timeout mode. I've tried restarting Octoprint several times, but no luck. I don't know that increasing the distance would help, since it seems like it would just delay the trigger by a bit. Watching the distance remaining value while a print is going, the number there continually ticks down as it prints, eventually reaching zero. Should this number be getting reset back to 50 (or set value) whenever movement is detected? Or is it not quite that simple?

maocypher, 2020-12-29:
Yes, you are right. The timer starts with the first movement.

For distance detection: after a movement is detected (value of the sensor changed) the distance is reset

Chrosus, 2020-12-29:
I will try again after my current print finishes. What I was seeing was a strange calculation where the distance remaining value would go down, then increase by a smaller amount, then go down again, and repeat this over and over until it reached zero, at which point it would pause for a filament change.

Chrosus, 2020-12-29:
I set the distance to 60 and tested it again, with the same result. Something interesting I noticed. When it draws the first test lines, the number goes to 67 and stays there for the two lines. However, when it starts to do the skirt and the actual print, the number changes to a fractional number and continually decreases. When I stopped the print, this was the distance value showing in the newly added side panel: 20.196029999999997

maocypher, 2020-12-29:
Yeah, the numbers are strange. Could you provide me the first 10 lines of your print? Then I would try it on my printer

Chrosus, 2020-12-29:
;FLAVOR:Marlin
;TIME:1821
;Filament used: 1.52947m
;Layer height: 0.2
;MINX:101.7
;MINY:103.2
;MINZ:0.2
;MAXX:122.165
;MAXY:122.8
;MAXZ:20
;Generated with Cura_SteamEngine 4.8.0
M82 ;absolute extrusion mode
; Ender 3 Custom Start G-code
M140 S60.0 ; Set Heat Bed temperature
M190 S60.0 ; Wait for Heat Bed temperature
M104 S160; start warming extruder to 160
G28 ; Home all axes
G29 ; Auto bed-level (BL-Touch)
G92 E0 ; Reset Extruder
M104 S210 ; Set Extruder temperature
G1 X0.1 Y20 Z10.0 F5000.0 ; Move to pre-heat position
M109 S210 ; Wait for Extruder temperature
G1 X0.1 Y20 Z0.3 F5000.0 ; Move to start position
G1 X0.1 Y200.0 Z0.3 F1500.0 E15 ; Draw the first line
G1 X0.4 Y200.0 Z0.3 F5000.0 ; Move to side a little
G1 X0.4 Y20 Z0.3 F1500.0 E30 ; Draw the second line
G92 E0 ; Reset Extruder
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
; End of custom start GCode
G92 E0
G92 E0
G1 F2700 E-4
;LAYER_COUNT:100
;LAYER:0
M117 INDICATOR-Layer0
M107
;MESH:xyzCalibration_cube.stl
G0 F6000 X120.45 Y121.95 Z0.2
;TYPE:WALL-INNER
G1 F2700 E0
G1 F1200 X102.55 Y121.95 E0.59536
G1 X102.55 Y104.05 E1.19071
G1 X120.45 Y104.05 E1.78607
G1 X120.45 Y121.95 E2.38143
G0 F6000 X120.85 Y122.35
G1 F1200 X102.15 Y122.35 E3.00339
G1 X102.15 Y103.65 E3.62535
G1 X120.85 Y103.65 E4.24732
G1 X120.85 Y122.35 E4.86928
G0 F6000 X121.25 Y122.75
;TYPE:WALL-OUTER
G1 F1200 X101.75 Y122.75 E5.48543
G1 X101.75 Y103.25 E6.10157
G1 X121.25 Y103.25 E6.71772
G1 X121.25 Y122.75 E7.33386
G0 F6000 X121.05 Y122.75
G0 X120.81 Y122.31
G0 X120.34 Y121.56
G0 X120.06 Y121.56
;TYPE:SKIN
G1 F1200 X102.94 Y121.56 E7.90327
G1 X102.94 Y104.44 E8.47269
G1 X120.06 Y104.44 E9.0421
G1 X120.06 Y121.56 E9.61151
G0 F6000 X120.34 Y121.56
G0 X120.34 Y105.192
G0 X119.899 Y105.192
G1 F1200 X119.308 Y104.6 E9.63934
G0 F6000 X118.742 Y104.6
G1 F1200 X119.899 Y105.757 E9.69376
[... 900 lines cut ...]
G1 F4200 X104.779 Y104.079 E217.34924
G0 F10500 X104.43 Y104.079
G1 F4200 X102.579 Y104.575 E217.41298
G0 F10500 X106.068 Y104.078
G1 F4200 X102.579 Y117.1 E217.86137
G0 F10500 X102.579 Y116.999
G1 F4200 X120.418 Y112.219 E218.47563
G0 F10500 X120.42 Y111.234
G1 F4200 X113.265 Y104.079 E218.81218
G0 F10500 X112.28 Y104.078
G1 F4200 X107.499 Y121.92 E219.42654
G0 F10500 X107.397 Y121.919
G1 F4200 X120.418 Y118.43 E219.8749
G0 F10500 X114.134 Y121.919
G1 F4200 X102.579 Y110.365 E220.41839
G0 F10500 X102.58 Y110.787
G1 F4200 X120.418 Y106.007 E221.03261
G0 F10500 X118.492 Y104.078
G1 F4200 X113.711 Y121.92 E221.64698
G0 F10500 X105.648 Y121.919
G1 F4200 X102.579 Y118.85 E221.79133
;MESH:NONMESH
G0 F300 X102.579 Y118.85 Z2.4
G0 F10500 X120.5 Y122
;TIME_ELAPSED:261.945602

Chrosus, 2021-01-05:
Did the gcode for the first 10 lines cause the same issue? To add detail, I used a heavily modified Ender 3 with an SKR 1.2 mainboard connected to a Raspberry Pi 3b+. Not sure if any of that matters however.

maocypher, 2021-01-05:
Hi, sorry for the open response. I can try tomorrow.
I had another small print (late x-mas present) running, but without issues.
Detection Distance was 50. I had some issues in the beginning, but it was a wrong leveled z-axis.

Did you check if you have underextrusion or slipping filament in the extruder?

Chrosus, 2021-01-05:
That was one of the first things I checked, both related to the sensor and the extruder itself. Didn't find any issues with that.

maocypher, 2021-01-06:
As promissed I ran the print. I ran into the same issue you deacribed with 100mm configured.

I am going to run it in Debug-Mode again and take a look if there is an error logged

maocypher, 2021-01-08:
Hi :) please try https://github.com/maocypher/Octoprint-Smart-Filament-Sensor/archive/1.1.5.1.zip I fixed a bug in the distance detection. This worked with your gcode

Chrosus, 2021-01-08:
Thank you very much! It seems to be working now.

closed 2021-01-08 by Chrosus

mylife4aiurr, 2021-01-13:
Im having a hell of a time on 1.1.4 with this same issue. Any reason 1.1.5 isnt released yet?

Gonna try 1.1.5 since link is here

Chrosus, 2021-01-13:
> Im having a hell of a time on 1.1.4 with this same issue. Any reason 1.1.5 isnt released yet?
> 
> Gonna try 1.1.5 since link is here

There is a link in a previous comment to 1.1.5.1 made my maocypher. Copy that URL and you can install this build by going to the plugin manager, selecting the option to get more plugins, and there is an option there to install by url. Paste url for 1.1.5.1 there and it should work.

maocypher, 2021-01-13:
Yes, the reason is, that I completly refactored the code and used rebase instead of merge for the first time.

There is already 1.1.5.2 from yesterday, because another big was found.

If everything is fine over the weekend, I would say I go for it and release it officially