maocypher/Octoprint-Smart-Filament-Sensor #25: GPIO pin numbering note in Readme?

zapdogster, 2020-12-29:
A suggested ReadMe addition for those not being familiar with RPi GPIO. I did not know the difference between BCM and BOARD mode numbering. I would have found something like the following in the GPIO section to a be good heads up for a newbie:

•	Be aware of the difference between  BCM and BOARD for the “Board Pin Mode” mode dropdown selection:
    o	BOARD mode numbering refers to the physical pin number of the GPIO connector
    o	BCM mode numbering refers to the pin function, by the "Broadcom SOC channel" numbering  - i.e. the numbers after the “GPIO” label
     o	Example on a RPi 3B: BOARD mode pin 23 == BCM mode pin 11 (i.e. GPIO11)

Thanks for the great plugin!

maocypher, 2020-12-29:
Thanks for the suggestion. I will extend the description soon :)

maocypher, 2021-01-08:
Extended description for the next release
https://github.com/maocypher/Octoprint-Smart-Filament-Sensor/tree/PreRelease

closed 2021-01-08 by maocypher