maocypher/Octoprint-Smart-Filament-Sensor #26: BigTreeTech sensor vs Octoprint Smart Filament Sensor

bulgaro80, 2021-01-02:
Hi all,

I have an issue with BigTreeTech Smart Filament Sensor and this plugin. After some analysis seems it's an issue (or any stupid config. issue created by myself....) related to the plug-in and not to the phisical sensor.

WHAT'S HAPPEN: when I start a print (even with BigTreeTech Smart Filament Sensor unplugged) everytime at the same point the printer move to pause (for unknown reason). Even if I change the Pin or the detection time, it stops at the same point (while it prints the skirt). here below how it's configured. As soon as I disable the plugin, everything works fine.

[image]

MY TOOLS: Raspberry Pi4, Octoprint 1.5.2 and Ender 3 V2 with Marlin.

Any idea on what can I do to solve the issue?
Many thanks,
Stefano

maocypher, 2021-01-02:
Hello Stefano, please try to increase the distance or the time.

From experiance you cannot use the recommend value from BTT. Try to increase the value in steps of 5mm or 5secs on a test print. FYI the value somehow depends on print speed and length. I am still trying to figure out why

bulgaro80, 2021-01-02:
Many thanks for your quick answer. Yep, I’ve increased the distance (30mm) and now it works (doesn’t stop anymore at the beginning of a print).
Now the last remaining issue is related to M600 command and Ender 3 V2 (it’s not natively supported). 😭

I’ve implemented the following procedure: https://youtu.be/P8Q3RmZWNzg , including the pause and resume GCODE and the issue is that after pressing resume in Octoprint, the nozzle go back to the previous position for axes Y&X but not for Z... so it starts to print in the air... (I’m not an expert in GCODE but seems an issue related to the variables in the GCODE commands... any idea about it?

maocypher, 2021-01-02:
This is my pause code run by Octoprint

```
G91 Z              ; Set to relative positioning mode
G1 Z15             ; Move hotend away from part by 15mm
G90 Z              ; Set to absolute positioning mode
```

And this is my resume code before the print continues

```
G91 Z              ; Set to relative positioning mode
G1 Z-15            ; Return to original position
G90 Z              ; Set to absolute positioning mode
G92;
```

If you move the nozzle afterwards you need to ensure it is in the pausing position again.

With G60 it could be possible to save the last position and restore it with G61, but I didn't try it yet

https://marlinfw.org/docs/gcode/G060.html

bulgaro80, 2021-01-03:
Finally I solved it! I realized that I had a plug-in within Octoprint called "Filament Change - M600 Rewriter (1.0.5)" and after M600 command, it was moving up Z to +80mm.... Now I disabled the plugin and it works!
Many thanks!!

Waiting for Version 1.1.5 to be released!

closed 2021-01-03 by bulgaro80