maocypher/Octoprint-Smart-Filament-Sensor #27: support Ir Filament Sensor

XSJFYZS, 2021-01-03:
Is it possible to support this Ir Filament Sensor?
Because it is cheaper, I don’t print transparent filament
I use klipper firmware, so can’t configure it
Thank you for your work
https://www.aliexpress.com/item/32976742501.html

maocypher, 2021-01-03:
Hi :) as I understand the IR sensor it is a simple NC-Sensor (normaly closed)

https://help.prusa3d.com/en/article/ir-filament-sensor-calibration-mmu2s_2245/

There are other Octoprint Plugins like e.g. did you already try this instead?
https://plugins.octoprint.org/plugins/filamentrevolutions/

XSJFYZS, 2021-01-04:
Very strange, the product description shows that it is an optical encoder, jam can be detected
This company does not sell Smart Filament Sensor in China
I don’t really understand the sensors used by this plugin
Is it two ordinary switches?
Like this? https://www.thingiverse.com/thing:3319350

maocypher, 2021-01-04:
No, the plugin in is for e.g. for Bigtreetech Smartfilamentsensor or sensors that send alternating signals that are changing regulary from 0 to 1 and back to 0 depending on the distance the filament passed through the sensor.

The sensor from BTT realizes this with two or or three wheela inside the sensor. These wheels are turned if the filament moves. If the filament is not moving anymore the signal is not changing anymore, but the print continues until the firmware or the plugin says too much distance was passed without a change in the filament

https://youtu.be/9gLSJIxiuwo

XSJFYZS, 2021-01-04:
No, I am asking about this https://plugins.octoprint.org/plugins/filamentrevolutions/
Does this plugin use BTT or two switches?

maocypher, 2021-01-04:
It uses one or two switches of the kind you send

XSJFYZS, 2021-01-04:
thank you very much

maocypher, 2021-01-08:
Is the other plugin working for you?

mylife4aiurr, 2021-01-14:
Got it working with IR, triggered by encoder wheel.
https://www.thingiverse.com/thing:4712966

info here #29

XSJFYZS, 2021-01-14:
> 它与红外传感器一起工作，由编码轮触发。
> https://www.thingiverse.com/thing:4712966
> 
> 这里的信息[＃29](https://github.com/maocypher/Octoprint-Smart-Filament-Sensor/issues/29)

Coooool！

maocypher, 2021-01-14:
I asked the community of the YT Channel to test the plugin with the sensor. Maybe they give it a try