maocypher/Octoprint-Smart-Filament-Sensor #30: GPIO not working in BCM mode for other plugins

fspralja, 2021-01-16:
Hi, after updating the plugin to 1.1.5.3 other plugins using bcm mode like enclosure plugin stop working, disabling the smart filament sensor resolves the issue for now. Any suggestions? ty

arathas, 2021-01-16:
I have the very same issue, me it with this plugin and PSU Control

maocypher, 2021-01-16:
Hi, could you please try to uncomment the line with
GPIO.cleanup?
Maybe this causes it. If yes, I can fix that

fspralja, 2021-01-17:
Yes, commenting out cleanup helped.
```
        def _setup_sensor(self):
        # Clean up before intializing again, because ports could already be in use
        #GPIO.cleanup()
```

other plugins have the cleanup as a web action (but probably a reboot cleans up as well):
enclosure plugin warning on that button:
>_"This will clear any GPIO configuration that might exist on your system, use with caution, it might break other plugins that use GPIO"_

Hobby4life, 2021-01-26:
This really should be an option in the plugin UI configuration.

Thegs68, 2021-03-22:
Hi
I was wondering where the init.py file is located. So that I can try and resolve the same problem.

fspralja, 2021-03-22:
Hi, for me the path was
```/home/pi/oprint/lib/python2.7/site-packages/octoprint_smart_filament_sensor/__init__.py```

griehsler, 2021-04-01:
Heads up on this one: I had exactly the same issue, Smart Filament Sensor breaking PSU Control. Commenting out `GPIO.cleanup()` fixed it.

@maocypher +1 for removing it officially. GPIO is shared across all plugins. Cleanup needs to be used with care as it might (and obviously does) undo setup done by other plugins.

allenrowand, 2021-04-10:
Commenting the line resolved the issues regarding interference with other GPIO configurations but stopped the sensor from working for me.

I realized the the side bar always showed the filament was moving (even when it wasn't) and I had a runout which wasn't stopped. Afterward I ran the sensor test and the sensor behaved as expected, but when I started the next print it didn't register filament movement and paused printing even though there was filament loaded.