maocypher/Octoprint-Smart-Filament-Sensor #31: Interference with other plugin

jtannenba, 2021-01-16:
Don't know if this is a problem or just miscommunication of documentation.  Have PSU set up and working just fine, then added the smart filament plugin and a sensor, now psu won't work.  the Pi doesn't toggle, it appears it floats.  The relay I'm using requires a low to turn off.  Both plugins are set to BOARD and PSU is on pin 12 the filament sensor is on 11.  If I disable the filament sensor, psu works again.  Should I set one or the other to different pins?

maocypher, 2021-01-16:
Hi, could you please try to uncomment the line with
GPIO.cleanup in init.py?
Maybe this causes it. If yes, I can fix that.
A restart is needed after this

jtannenba, 2021-01-16:
Where would I find this?

jtannenba, 2021-01-16:
Before I saw this was trying to see if it would work if psu was board and sensor was bcm.  the test seemed to work and psu worked, so tried a print with a short piece of filament to try to see if it would detect it and the nozzle parked without running out??? I Disabled it again and retried the print which completed fine.

I do all my printing through Octoprint.  Love using the RaspberryPi's.  I have them for file server, and Ups monitors in my house.
I hope this gets fixed so I can re-install the sensor.

ChristianH99, 2021-01-17:
Removing the line GPIO.cleanup fixed the problem for me.

Follow these steps @jtannenba:
Login on the pi via ssh
cd ~/oprint/lib/python3.7/site-packages/octoprint_smart_filament_sensor
nano __init__.py
write an # in front of GPIO.cleanup
Save with ctrl + x then y and enter
Reboot pi

jtannenba, 2021-01-17:
I will try this later today.  I didn't realize how to get at it.  What is the default login for the pi?

maocypher, 2021-01-17:
Hi thanks for the feedback, I am going to fix this in the next days.

Default login shoul be username pi and password raspberry

jtannenba, 2021-01-17:
I'm getting access denied.

maocypher, 2021-01-17:
Maybe z and y is changed on the keyboard? Sometimes by default the english keyboard is used. I have a german layout so y is a different key

jtannenba, 2021-01-17:
Duh, just remembered, I changed the password when setting it up.  Ok I'm in, but I had removed the plugin so the package is not there.  I'll experiment later today or tomorrow, when I get some time.  Thanks for your help.

jtannenba, 2021-01-17:
Thank you, got it.
Joe

 
 

Default login should be username pi and password raspberry

Drzet, 2021-01-17:
PSU plugin has issues, I don't think your problem was caused by adding filament sensor.
On quite a few occasions PSU would "forget" the pin or switch to gcode control altogether.

maocypher, 2021-01-17:
@Drzet thanks for the information, do you have an issue number I can reference to?
In that case I wait if there are others with the same issue, but other plugins. If not, this issue is goin to be closed in a few days

ChristianH99, 2021-01-17:
Actually PSU had the same issue some time ago: https://github.com/kantlivelong/OctoPrint-PSUControl/issues/5

There although was an issue a few months ago that PSU didn't store the GPIO pin correctly after leaving setting, but this is solved and I never had problems with PSU again.

jtannenba, 2021-01-17:
I've been usng PSU since the day after I got my printer..The Monday after Christmas, and it's been working flawlessly. Getting ready to try the above fix. I hope you meant __init__.py and not init.py?

jtannenba, 2021-01-17:
Ok, PSU working again and the movement test works, but, I put a short piece of filament into the sensor and extruder (removed the bowden tube going to hotend-The sensor is in front of the extruder) and told it to print.  It ran the filament completely out and nothing happened.  So, what is supposed to happen?  What did I miss.  I didn't mess with the firmware at all.  I'm using Marlin's edender 3 v2 stock firmware.  Since I'm using the Pi, I still have this:Enabled Smart Filament Runout Sensor (Add M412 S0 in your Start G-Code if you do not have a filament sensor). in my start code.  I believe that is only for when it is hooked into the main board?  I left the settings default except changed to distance and left at 15 mm.

maocypher, 2021-01-18:
Hi, what should happen, if the sensor detects jams or runouts it should pause the print.
After setting everything up, a configuration of the distance or Timeout might be necessary. You do not need to follow the instructions from videos about BTT sensor on the board, because this is totally different. Just the instructions in Readme.md

The plugin is implemented in python, because this is how the plugin architecture from Octoprint works. If you are interessted in more details, here is the tutorial for plugin: https://docs.octoprint.org/en/master/plugins/gettingstarted.html

jtannenba, 2021-01-19:
Monday:  Saved a print when I ran out of filament.  Got it going and when the print finished PSU did not power printer down, so manually clicked power off and no go. Still something between them?

maocypher, 2021-01-19:
If everything is fine for you :)
A feedback would be good, if you are running everything with or without GPIO.cleanup

I am not using PSU Control, so a short feedback from your side is necessary to decide

jtannenba, 2021-01-19:
So far, so good.  I did the edit and it saved a print last night.

jtannenba, 2021-01-20:
When starting prints, it parks the nozzle, I see nothing wrong.  Once printing, no problems

maocypher, 2021-01-20:
Sometimes higher values are needed. This depends e.g. on print speed. Unfortunately I currently have no solution to automatically find out the best values

jtannenba, 2021-01-20:
It just parked again.  This is frustrating.  Printed most of yesterday just fine, now when I'm trying to tune in ABS, it keeps doing this.
[octoprint.log](https://github.com/maocypher/Octoprint-Smart-Filament-Sensor/files/5844582/octoprint.log)
I included the octoprint log.  Looks like some of it is in German?

maocypher, 2021-01-20:
My plugin is completly coded in english. If there are german parts ins the log, this in not entered by my plugin.

For connection test, everything looked good as far as I could see.

``` LOG
octoprint.plugins.smartfilamentsensor - INFO - Error: Disabling filament sensors.
```

This line says that an error event was thrown from Octoprint that stopped the print. In this case the plugin is stopped.

``` LOG
2021-01-20 07:45:17,769 - octoprint.util.comm - ERROR - Unexpected error while connecting to serial port /dev/ttyUSB0, baudrate 115200 from hook default: SerialException: '[Errno 2] could not open port /dev/ttyUSB0: [Errno 2] No such file or directory: '/dev/ttyUSB0'' @ comm.py:_open_serial:3670
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/util/comm.py", line 3670, in _open_serial
    settings().getFloat(["serial", "timeout", "connection"]),
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/util/comm.py", line 3644, in default
    serial_obj.open()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/serial/serialposix.py", line 268, in open
    raise SerialException(msg.errno, "could not open port {}: {}".format(self._port, msg))
SerialException: [Errno 2] could not open port /dev/ttyUSB0: [Errno 2] No such file or directory: '/dev/ttyUSB0'
```

Here it looks like an error on the USB port caused this error.

For different filaments it might happen that the sensor behaves different. For me TPU was detected the best so far

jtannenba, 2021-01-20:
Do you think I should reload Octopi?  I only have 2 plusgins, yours and PSU.   Ater the reboot, it seems to be going ok. Print is over and hour into it.

maocypher, 2021-01-20:
Maybe it is a good idea to restart the RaspberryPi completly. Not only OctoPrint. In most of the cases a reboot a OctoPrint helps when it is somehow "messy", but due to the USB error from the USB port - it could also be you just unplugged the cable - but I would restart it.
Maybe the Pi is already running a long time - like mine regularily ^^"

jtannenba, 2021-01-20:
I had a pi 3b as a fileserver.  Ran from when they came out till I got a 4. Only down times were when power went out (it was monitoring the UPS and powered other fileservers down before it shut down.  They can run long times.  That USB error, may be one of the times I powered the printer down.

maocypher, 2021-01-21:
If you see a pattern what could cause the strange behaviour, let me know. Then I can try to search for something

jtannenba, 2021-01-21:
Ok. Busy the next few days.

jtannenba, 2021-01-24:
ok...I switched off PSU to change filaments and run an Esteps on the new one.  Switch it back on (PSU is on when the Octopi's pin is low, so I put a swith in the relay box to disconnect when I want to use printer from control panel)  After starting a 16 hr print,  at about 13 min in it paused, parked nozzle and backed out filament.  I told it to print and it seems like it is. (I hope, long print).  Seems like it pulls this after working the printer from the control panel and going back Octoprint.

jtannenba, 2021-01-29:
Printing with ABS.. 5 hr print. Prior to starting, rebooted Pi.  2 hrs into a 5 hr print, the nozzle parked.  I got it going, but it messed up and parked again, this time on continue it was messing up the print and I had to kill it. Log was too big, I hope I have the revelant portion.  Apparently the full screen plug it goes wild in the log.  Also after loading the new version of Octopi, i didn't set the time to local.
[new 1.txt](https://github.com/maocypher/Octoprint-Smart-Filament-Sensor/files/5890749/new.1.txt)