maocypher/Octoprint-Smart-Filament-Sensor #34: Movement still detected after runout.

chickenwoman, 2021-01-21:
Hi, thanks for creating this. I'm using distance mode set to 15mm with a GreenTreeTech smart sensor. I've tried 1.5.2 and 1.5.3. When printing the movement numbers change reflecting movement, however when it runs out of filament the numbers keep changing instead of counting down and the print doesn't pause. The sensor properly detects movement and no movement in the test script. It did work when I first installed 1.5.2. I did a 20" test print and tested it over and over and it worked great. However after rebooting the system and starting a new print it stopped working. I then upgraded to 1.5.3 and had the same issue.

chickenwoman, 2021-01-21:
I've been using the same pause command of M25. When it does work the distance measurement slowly goes down. What's happening when it malfunctions is the distance counter keeps blinking 15mm.

maocypher, 2021-01-21:
If it is blinking I would assume that the signal is continously changing between 0 and 1.
Maybe you could change the log to debug mode and see how the values really changed

chickenwoman, 2021-01-21:
What would cause the sensor to keep changing between 0 and 1? When filament is moving the distance measurement bounces around between 15mm and 6mm, when it's out of filament it stays on 15mm while quickly blinking another number. I looked in the log and I don't see it reporting 0 and1a. Perhaps I'm looking at the wrong log. I tried it with timeout mode and it reported this back:
TypeError: printer_change_filament() takes exactly 1 argument (2 given)
2021-01-21 15:31:41,913 - octoprint.cli.server - ERROR - Uncaught exception
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_smart_filament_sensor/filament_motion_sensor_timeout_detection.py", line 50, in motion
    self.callback(True)
TypeError: printer_change_filament() takes exactly 1 argument (2 given)
2021-01-21 15:31:41,921 - octoprint.cli.server - ERROR - Uncaught exception
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_smart_filament_sensor/filament_motion_sensor_timeout_detection.py", line 50, in motion
    self.callback(True)

maocypher, 2021-01-21:
In the logging section you can activate Debug level instead of info. Then you can see every change of the remaining distance in the log

For timeout detection, I take a closer look on this error

chickenwoman, 2021-01-21:
Thanks, sorry to be a pain. There's different options for debugging, do I want to choose events or access?

maocypher, 2021-01-21:
[image]

Like this

chickenwoman, 2021-01-21:
Thanks, I wasn't looking hard enough. I set the debug log. I'm not seeing the 0 and 1s but it does show the movement count. I'm attaching my log. It worked twice triggering the M25, but the 3rd time it didn't, it kept counting. I canceled the job shortly after it wouldn't trigger. 
[octoprint (13).log](https://github.com/maocypher/Octoprint-Smart-Filament-Sensor/files/5852388/octoprint.13.log)

chickenwoman, 2021-01-26:
Just a follow up. I think it was electrical interference and not the plugin that was causing my issue. I have a large Modix printer with a long shielded cable run. I shorten the BigTreeTech cable and plugged it into my shielded wire but a small section of the cheap wire was near my hot end wire. Seems like that was enough. I'll keep you posted if I have issues again but it's ok to close the case. Sorry for any extra work I caused you. I just finished a 4 day print and your plugin saved me several times. Thank you for creating it.

maocypher, 2021-01-26:
Hi thanks for the information and the feedback.

I leave it open a few more days and if everything is fine I close it then

cwgismo, 2021-02-21:
I am using the plugin and I am having the same issue. your test script appears to be functioning correctly and detects filament movement from the sensor properly. However when using the plugin during a print it appears to read filament moment when there is none and when the filament is moving through the sensor it will randomly detect the filament stopped moving and sends the pause command even though filament is still moving. I am using a shielded cable and I have tested the sensor using a python script I found on the web (https://github.com/kgbplus/gpiotest)to test if the sensor is reading stray signals and it is not. I will update with a log file from Octoprint as soon as I can.

Hofei90, 2021-02-24:
Hello,
I have the same error as described above.
The filament has run out (deliberately cut off), but the distance value in the display of the Smart Filament Plugin continues to reset to the initial value.
The printer is not put into pause mode (M600) and continues to print.
Also for me, the manual sensor test works fine and as expected. However, during printing it is not detected when there is no more filament.

To test if there are disturbances or similar on the signal line, I ran a small gpiozero script with the following content:
```
from gpiozero import Button
from signal import pause

def say_hello():
    print("Hello!")

def say_goodbye():
    print("Goodbye!")

button = Button(24)

button.when_pressed = say_hello
button.when_released = say_goodbye

pause()
```

No false detection is detected here.

My version number:
OctoPrint 1.5.3
Python 3.7.3
OctoPi 0.17.0

Smart Filament Sensor 1.1.5.3)
in addition the following settings

BCM Mode
Distande dettection
M600 – Filament Change 
Mtion Sensor GPIO Pin 24
Detection distance: 20mm

Log: [octoprint.log](https://github.com/maocypher/Octoprint-Smart-Filament-Sensor/files/6035932/octoprint.log)

hawktrainer2222, 2021-03-17:
Hello,

I am having a similar/same issue. The test in the plugin settings window shows detection and filament movement properly. When printing the status bar message does not reflect the actual detection state, and if filament runs out or stops no gcode command is issued (for both timeout and distance detection).

I am using Klipper, but I have a macro that creates an M600 command, which works fine when I issue M600 in the terminal, so I don't think it's an issue with Klipper compatibility.