maocypher/Octoprint-Smart-Filament-Sensor #35: Filament runout issue.

Hobby4life, 2021-01-26:


maocypher, 2021-01-27:
Did you try e.g. M1 instead?

Hobby4life, 2021-01-27:
I dont understand.. can you explain what you mean with M1 command?

maocypher, 2021-01-27:
In the plugin you can change which pausing command is send.
But all you want to use must be configured in OctoPrint as pausing commands. Otherwise the print won't be paused properly

Hobby4life, 2021-01-27:
Octoprint is sending a M600 fine,, the printer is peforming a filament runout procedure then parks. .. but in octoprint I can not press a button to resume again,, While selecting the virual printer, all is behaving as it should. I get the resume button.
Normally you press resume on the printer,, but the display on the ender is a DWIN display, and does not show anything else then temperatures when printing via USB.
Best is to have a box where you could define your own GCODE procedure.

Hobby4life, 2021-01-30:
I have succesfully configured octoprint with the GCODE script for pausing and resuming behaviour i like to see.
So when pressing pause and resume in octoprint it performs my GCODE scripts.

Although I can not seem to get this plugin to perform a "octoprint" pause.. I have tried all Pause command's of the plugin, but none initiate the scripts..

Hobby4life, 2021-01-30:
Ok after a full restart of octoprint and printer.. M1 seems te work.
But now I get a Red message box in octoprint saying M1 command is not sent and is blocked..

Hobby4life, 2021-01-31:
I have Implemented a real octoprint pause and cancel command, and all is working perfectly now.

I have added 2 commands:

```
                    <option value="@pause">{{ _('OctoPrint - Pause print') }}</option>
                    <option value="@cancel">{{ _('OctoPrint - Cancel print') }}</option>
```