maocypher/Octoprint-Smart-Filament-Sensor #42: Big tree tech sensor not working with plug-in? Config?

dretful, 2021-02-08:
I have installed the big re-elect, and when printing without the plugin I get filament timeout messages and it detects movement or lack of it fine.
When I enable the plugin the detection seems to be a bit random, either with time or distance, had to set thresholds very high like 100 secs 100 mm. It might be detecting noise?
The prints are getting stopped by the plugin when it decides the filament is not moving - even though it is printing fine at the time.
So my questions. 
1 The big tree tech does not physically to the raspberry pi pins. So what should the Gpio be set to in the plugin config, and why? I have tried -1 and 11. Both cause the prints to be stopped, not her seem to respond to the test.
2 when I use the test mode both the message “filament is not moving “ and “filament is moving” texts show initially. So how do I tell if the test is working?. The UI for the test confuses me.
Help please?

maocypher, 2021-02-08:
Hi :)
You are right the description in the frontend of the connection test is misleading.

**Filament is moving: filament is not moving** means there is no movement detected.
**Filament is moving: filament is moving** means it is moving

You must connect the sensor physically to the raspberry. Therefore the signal pin could be e.g. GPIO 17 in BCM oder Pin 11 in Board mode. -1 is the default value to deactivate it right after a fresh installation.

After setting this up try a connection test. As soon as it shows **Filament is moving: filament is moving** when you move the filament manually in the sensor, everything is set up correct.

The next step could be a little tricky. You could 
- start with a low value 15mm and turn is up till no false detections are recognized anymore
- start with a high value and monitor the sidebar what could be a suitable value

I am going to improve this in the future. I am not sure why the sensor is so unreliable, but it is not sending signals every 4-7mm as told in many videos.
It depends on many factors like the filament type (ABS, PETG, PLA, TPU), print speed, length of printed lines, how much retraction is needed and so on.

I run my set up with 20-30mm sometimes at 60mm for bigger prints. Others told they need 100mm. Software cannot fix the unreliability of hardware

dretful, 2021-02-08:
Thanks Anni.

I don’t know how to connect the bigtreetech to the raspberry pi.  Its connected directly to the port on the board and it seems to work already, detecting the messages (somehow) when it decides that the filament isn’t moving.  I think that’s magic! 

Here’s an article that explains how I’ve connected it.
(1) Ender 3 V2 filament runout sensor how to? : ender3v2 (reddit.com)

And this tells me more about how to connect it to a pi I guess.  Its just that I already ran all my cables doh!

So tuning will come later maybe but I guess you’ve answered my question, that unless its attached to the pi it wont work.  

Instead it seems the sensor just sends an M600 whenever it thinks it needs to.  And the Octoprint UI picks that up anyway.  Maybe it could handle it differently with alerts  I don’t know.

So I think that’s the end of the road for this enquiry except perhaps to askj that in your plugin you say that if you connect it that way I did  you don’t need the plugin?
Thanks for all your work

Tony

From: Anni L.
Sent: 08 February 2021 18:08
To: maocypher/Octoprint-Smart-Filament-Sensor
Cc: dretful; Author
Subject: Re: [maocypher/Octoprint-Smart-Filament-Sensor] Big tree tech sensornot working with plug-in? Config? (#42)

Hi :)
You are right the description in the frontend of the connection test is misleading.
Filament is moving: filament is not moving means there is no movement detected.
Filament is moving: filament is moving means it is moving
You must connect the sensor physically to the raspberry. Therefore the signal pin could be e.g. GPIO 17 in BCM oder Pin 11 in Board mode. -1 is the default value to deactivate it right after a fresh installation.
After setting this up try a connection test. As soon as it shows Filament is moving: filament is moving when you move the filament manually in the sensor, everything is set up correct.
The next step could be a little tricky. You could
• start with a low value 15mm and turn is up till no false detections are recognized anymore
• start with a high value and monitor the sidebar what could be a suitable value
I am going to improve this in the future. I am not sure why the sensor is so unreliable, but it is not sending signals every 4-7mm as told in many videos.
It depends on many factors like the filament type (ABS, PETG, PLA, TPU), print speed, length of printed lines, how much retraction is needed and so on.
I run my set up with 20-30mm sometimes at 60mm for bigger prints. Others told they need 100mm. Software cannot fix the unreliability of hardware