maocypher/Octoprint-Smart-Filament-Sensor #45: BTT Sensor stopping beginning of print

jamie45678, 2021-02-16:
I installed the latest version of the plugin. I have it setup on a RPI4 it will show the filament is moving then run the number to negative and then pause the print right at the start. I will a fake filament change. After that it never showes the filament as moving, but the remaining distance number will change randomly and it will finish the print.

ryancasler, 2021-02-19:
I had something similar happen when I had the sensor configured for timeout.  Once it begins printing, the timer starts.  So, that includes warm-up time.  I believe that is why distance is the preferred method.  Although, I have not gotten that to work reliably yet.

jamie45678, 2021-02-19:
I did have it on distance. I tried it both ways. I plugged it into my printer and updated the firmware. With the latest update it works great plugged into the printer. It has 2 way communication with octoprint now.

Patbonamy, 2021-12-30:
Hello,
I have the same issue, do you solve it?