maocypher/Octoprint-Smart-Filament-Sensor #48: M0, M1, M25, M226, M600, M601 are not good pause commands

Ruissilva, 2021-03-17:
Hello,

I have a ender 3, with the latest bugfix version of marlin. My smart sensor is working fine, except for the part that i have i have to choose a pause command. I tested all the options throw the octoprints plugin and these are my conclusions:

M0, M1 - Give a error message.
M25 - Enters in pause and it is not possible to remove pause. 
M226 - Nothing happens. 
M600 - Switch filament. It is not a pause command.
M601 - Unknown command.

So, the way I see it, i have to connect the sensor to the motherboard. Has to work that way, but the problem is that octoprint is blind that way. If the print goes to stop by the motherboard, octoprint will not recognize it, and I will not get a warning that the printer is stopped, for example, by telegram,but at least the printer goes to pause. 

So, my question is, since I just tested this sensor for the first time, am I missing something?

Greetings from Portugal,
Rui Silva

Ruissilva, 2021-03-19:
Hello again, I had an ideia that might solve my problem. I want help creating a Gcode, M601.

Something like:

M601{
Go to pause position
LCD: "Do filament change? Yes or No"
If Yes: Do M600
If No: Resume Print.
}

This will help a lot of people using the new smart filament sensor, because this way we dont have to change the filament if we have a clogged nozzle or a spool jam. 

Can you please help?

Thank you in advance

Prizmagnetic, 2021-04-03:
Why can't you just add M600 to octoprints pausing commands?  And then you can change filament, or not, or whatever, and then hit resume in octoprint when you are ready

Royrdan, 2022-01-15:
I added a pull request and added a @pause option in one of the templates files. Works for Octoprint then I could set any GCODE in the Octoprint pause commands. You can probably ad whatever you want as a pause option though.
https://github.com/maocypher/Octoprint-Smart-Filament-Sensor/pull/68