maocypher/Octoprint-Smart-Filament-Sensor #50: Smart filament plugin in conflit with other plugins that affect gpio pins

Ruissilva, 2021-04-01:
Smart filament plugin is in conflit with other plugins that affect gpio pins.

I tested Octorelay, PSU Control and Enclosure and they stop working well. The smart filaments seems to work fine, but the others don't. They lose their settings or cannot load their settings at start. The issue is more visible in octorelay, because it cannot load the icon images in the nav bar, and appear R1,R2,R3,R4,R5,R6,R7 and R8. I can only get it to work if I go to the settings, change the gpio pin number to a wrong one save, then go back to the menu and choose the write gpio pin. All seems to work ok(still testing) until i turn off and on octoprint.

allenrowand, 2021-04-02:
Having the same issue; with this plugin enabled, I can't control other GPIO pins through the Enclosure plugin. Disabling this plugin restores Enclosure plugin function.

With both plugins enabled if I trigger one of the Enclosure plugin actions I get this error in the log:
2021-04-02 10:39:27,619 - octoprint.plugins.enclosure - WARNING - An exception of type RuntimeError occurred on write_gpio when writing on pin 10. Arguments:
('The GPIO channel has not been set up as an OUTPUT',)
2021-04-02 10:39:27,665 - octoprint.plugins.enclosure - WARNING - An exception of type RuntimeError occurred on log_error. Arguments:
('You must setup() the GPIO channel first',)

Please let me know if you need specific tests done or other logs, thank you.

stefanelul2000, 2021-04-28:
Issue is still present. I only have the Smart Filament Plugin installed and no other plugin that controls GPIO pins. If I go in the settings menu, it reports as being functional, but if I start a print, it stops working and it continuously pauses my prints. I am available for tests and debugging if needed