maocypher/Octoprint-Smart-Filament-Sensor #60: Connection test button wont work

renegadecitizen, 2021-08-11:
Connection test button wont work
[image]
The button wont work
Ive tried cleaning the data and reinstalling the plugin but it still doesn’t work
Can you please help me?

renegadecitizen, 2021-08-11:
I just tried chrome and something it doesn’t work

stoneblock, 2021-09-13:
I had the same issue 
- I tried safari 12.1.2 (OSX 10.12.6 Sierra (older system)) and the button don't work
- because of your post I tried Firefox 92.0 and voila it worked fine the button is alive
maybe this helps for now

stoneblock, 2021-09-13:
crazy - after the first test on firefox are worked fine - now it works under safari too

renegadecitizen, 2021-09-13:
Odd, I uninstalled the plugin and went back to the stock configuration 
I’ve moved on to otherthings. 
so I can’t test it sorry

VirtualMike, 2022-01-21:
Does someone have a screen shot of what it is supposed to look like? I'm doing some testing and I can't get it to work either. Only tried Chrome, Edge and Chrome on a Windows PC and Chrome and Safari on an iPad.