maocypher/Octoprint-Smart-Filament-Sensor #65: Your plugin killed my whole Octopi

awiltschek, 2021-10-10:
Thnx for destorying my octopi ! After saving settings my pi never came back and hang up in endless loops.
So i can reinstall and setup from scratch again ! well done

jessearmstrong, 2021-11-29:
This sounds pretty unlikely.

ryancasler, 2022-01-18:
Yeah, I never had that problem with this plugin.

awiltschek, 2022-01-18:
nice but doesnt change the fact.
thats why i never will use it again after total reinstallation

ryancasler, 2022-01-18:
Then there is nothing for anyone to do to try and address the problem even if there was one.

VirtualMike, 2022-01-21:
I think they should get a complete refund of the money they spent on this plugin...oh wait...these people are doing this for FREE on their own time to try to help you, the user that can't write (or chooses not to write) their own plugin. huh.

awiltschek, 2022-01-22:
> I think they should get a complete refund of the money they spent on this plugin...oh wait...these people are doing this for FREE on their own time to try to help you, the user that can't write (or chooses not to write) their own plugin. huh.

then it should not pop up in octoprint or marked as experimental or non-professional. then everybody knows that there are respective risks of usage.

aldaw, 2022-04-08:
You could at least try to explain, what you have done exactly and what setup you have?
Maybe then one could reconstruate the problem and solve it.

But you are not warning at all, you are just bashing meaningless to people who did something for free.
And their intention was to make something not brake something... no one killed your pi on purpose.
You did on your own by that meaning.. if you cut your self with a sharp knife will you blame the manufacutre?