maocypher/Octoprint-Smart-Filament-Sensor #66: Power Source

DCib, 2021-12-21:
Trying to figure out if I can use this app before buying the smart filament sensor. 

Currently have my ender 3 pro connected to octoprint that has a screen attached. Problem is screen uses most of the GPIO connectors on the Pi 4b, so I don't have access to power thru the GPIO. 

Question is can I use 5v power from the USB-A connector on the pi to power the filament sensor and just use either GPIO pin 29, 31 or 37 for the signal wire ? My plan is to sacrifice a USB-A cable and tap into the power and ground wires

If I cant use 5v USB-A power can I then use a USB-A step down converter that converts 5v to 3.3v and get power and ground from that ?

ryancasler, 2022-01-18:
NO!   Absolutely not.  If you run this with 5v power you are going to get 5v out on the Signal pin and that will fry your RPi.  The GPIO on the Pi is on 3.3v tolerant.  So you have to supply the sensor with 3.3v ONLY.
Which screen are you running on top of your RPi?  Before trying to do that with the USB power, I would suggest using a gpio "doubler" like [this one](https://www.amazon.com/Connectors-Raspberry-40-pin-Expansion-RAS-GP02/dp/B07MCW4KCM) or something similar.  That is going to be a lot easier than trying to use a stepdown converter because I believe the USB power is isolated from the GPIO so I'm not even sure the ground on the USB would work with GPIO.  You're going to have to have at least 1 GPIO for signal anyway.

DCib, 2022-01-18:
Thanks for that info, this is the case and screen I'm using.

https://www.amazon.com/gp/product/B07XBVF1C9/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

I don't think I can use that type of GPIO doubler since it would make the screen stick up higher then the case. 
seems like I might not have too many options to use the smart filament sensor unless I decide to change screen and case of my pi.

VirtualMike, 2022-01-22:
Yea, that's a shame. It's got a huge GPIO connector stealing all the 3.3v lines. You might not be able to use it with that. What does the display do for you with OctoPrint? Don't you just use the web interface? All of my printers have a small case under or on the side of the printer to keep the Pi as much out of the way as possible.